libremesh/lime-packages #327: replace luci with lime-app

aparcar, 2018-02-13:
Can we collect here what's needed to replace luci in the default installation with the (significatly?) smaller lime-app?

Missing features:

- [ ] set admin password
- [ ] add ssh key
- [ ] install sysupgrade https://github.com/libremesh/lime-packages/issues/325
- [ ] change basic wifi settings

p4u, 2018-02-18:
For each network interface, define the protocols to use (WAN, LAN, BMX, BAT-adv, etc.)

Raw edit of /etc/config/lime via web interface (just for advanced users but might be useful)

aparcar, 2018-02-18:
@gmarcos87 if you like to work on that I'm happy to help with the ubus backend

nicopace, 2018-03-21:
@aparcar it is important to notice that lime-app was not meant to replace luci but to include the features that non-technical people require.

In that regard, I guess it is featureful as it is :)
My take would be for it to replace luci in the default image, and allow the hackers to add luci if desired.

ilario, 2018-12-08:
@nicopace @gmarcos87 can you update the list with what's ready and what is not wanted?

ilario, 2019-10-04:
I just tested lime-app (see also #609), it is very cool and looks powerful but I cannot yet use it at 100%.

From the list in the first comment, what is more important and still missing is (in my opinion):

* set admin password (in my case, extremely dangerous as the WAN port of my router is directly accessible from the internet, repeating the [issue we had more than 2 years ago](https://lists.libremesh.org/pipermail/lime-dev/2017-January/000827.html))
* add ssh key
* 2.4 GHz distance
* 5 GHz distance
* Country code
* raw edit of /etc/config/lime for advanced users

More things that are needed are:

* a way to have lime-app on HTTPS
* the notes set in lime-app should be printed also in the banner shown when connecting via SSH (see #611)

Things that should be working but seem not working to me:

* ubus-lime-batman-adv: should it add some information about Batman-adv in lime-app? I cannot see where
* ubus-lime-groundrouting: should it add a way to configure ground routing? I cannot see where
* ubus-lime-openairview: depends on fft-eval which is not present in 18.06 releases so I suppose will work once we base a release on OpenWrt 19.something
* location: I cannot move my node from the middle of the ocean, neither with Firefox nor with Chromium
* metrics: makes me wait forever with "Measuring links" (my node is directly connected to the internet with a public IPv4 on its WAN port)
* admin (accessible after setting the root password via SSH): changing IPv4 or changing "station name" makes me wait for then getting back to a login page where my password is always wrong, reloading the lime-app and checking for the IPv4, it did not change.

ilario, 2019-10-09:
lime-app has been released (see #615) so most of my comments above are likely outdated

ilario, 2019-12-01:
I suppose that some **community-wide settings** (like the IP network domain, main AP ESSID, Country code, AP open or WPA2-PSK, attended-sysupgrade, per-band AP or mesh modes) should be configured **during the network creation process in FirstBootWizard**.

On the other side, lime-app should configure just the node specific things (e.g. root password, per-radio distance parameter, manual IP, hostname).

ilario, 2020-01-04:
A [user just asked](https://lists.libremesh.org/pipermail/lime-users/2020-January/001565.html) if there could be an option for uploading a new firmware version. Makes sense for me.

sanadora, 2020-03-25:
Hello @ilario! I'm the one from the last email asking for an option to upgrade the firmware. Again, thanks for your answer. :smile: 

> it is important to notice that lime-app was not meant to replace luci but to include the features that non-technical people require.

I read that comment from @nicopace about one app not being the replacement for the other, and that both are complementary. It makes sense to me. 

Also, while the discussion evolves about what things of **luci** should be incorporated into **lime-app**, 
I think that I might be able to contribute with the documentation on the [Compiling LibreMesh from source code](https://libremesh.org/development.html) section, to clarify the way to install and use **luci** as an optional package, and the objectives of both apps.

What do you think?

ilario, 2020-03-25:
Maybe we could add a new page with description of optional packages (e.g. LuCI, 
prometheus-node-exporter-*, iperf3, pirania, tinc, wpad-mesh-wolfssl...), and link it from the development page?

aparcar, 2020-05-05:
I suggest removing this as an release blocker. Me and @dangowrt are looking into getting LiMe to work on 19.07, however we will not touch anything web interface related (because we can't). What do you think?

ilario, 2020-05-05:
> I suggest removing this as an release blocker.

Considering that LibreRouter people are already using lime-app in production on LibreRouter-os since long time, it makes sense. I remove the release-blocker tag. The most important features have already been implemented, for example the admin password setting is being addressed in #701.

> Me and @dangowrt are looking into getting LiMe to work on 19.07

That's awesome!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!

spiccinini, 2020-05-08:
> I suggest removing this as an release blocker. Me and @dangowrt are looking into getting LiMe to work on 19.07, however we will not touch anything web interface related (because we can't). What do you think?

Great! Yes I also think it is not a release blocker.