libremesh/lime-packages #332: lime-proto-bmx{6,7} don't announce gateway if same as bmx connection

aparcar, 2018-02-16:
If a node is connected on `wan` to a bmx network it will also use `watchping` on that interface. This leads to a situation where the node announces itself as uplink `/etc/watchping/wan-ok.d/bmx{6,7}-gw`) only having uplink thanks to a different (real) uplink within the bmx network.

What could be a solution for that? Not announcing bmx uplink when uplink via a bmx connection on wan?

dangowrt, 2018-02-16:
> What could be a solution for that? Not announcing bmx uplink when uplink via a bmx connection on wan?

Not installing lime-hwd-openwrt-wan...?

Re-using the interface names 'lan' and 'wan' has lead to confusion in
the past because it is not clear whether they are just re-used names
(such as when speaking bmx6/7 on an interface called 'wan' which
then certainly isn't used for 'wan' in the sense of the word).

aparcar, 2018-02-16:
> Not installing lime-hwd-openwrt-wan...?

I think it's a default package

dangowrt, 2018-02-16:
https://github.com/libremesh/lime-packages/issues/331

p4u, 2018-02-18:
IMO bmx should not run on WAN interface. By default it is like this (it runs on all network interfaces of the node) but it might be more convinient to use a different interface name.

This is the kind of section created by lime-hwd-openwrt-wan.

```
config net 'lm_hwd_openwrt_wan'
	option autogenerated 'true'
	list protocols 'ieee80211s'
	list protocols 'wan'
	list protocols 'bmx6:0'
	option linux_name 'eth0'
```
So if a routing protocol is configured on such interface, it is Ethernet and it's not a bridge,  lime-hwd-openwrt-wan should create a macvlan interface in /etc/config/network for runing only bmx{6,7} or whatever other routing protocol.

Then eth0 would be "wan" (so dhcp client) and eth0.foo would be the bmx{6,7} macvlan inteface only for routing packets.

@G10h4ck is the architect of lime-hwd-openwrt-wan, let's see what he has to say.

G10h4ck, 2018-02-19:
I am not sure I have understood well the situation,but it seems to me that not enabling other protocols on wan by default, is the simplest solution

aparcar, 2018-02-20:
Is that already the default or do we need to change something (all lime-protos?) for that?

G10h4ck, 2018-02-20:
In openwrt_wan.lua lines 28-37 and related is where this thing happen and just deleting those lines and doing some modifications in the surrounding lines should be enough to change this unwanted default behavior.

G10h4ck, 2019-02-16:
If i understand correctly this happens only if the wan of a libremesh router is connected to the lan of another libremesh router, this is something one should definitely not do.

closed 2019-02-16 by G10h4ck