libremesh/lime-packages #334: documentation

bitchecker, 2018-02-19:
hi,
the only documentation present is that of presentation that you see on the [site](https://libremesh.org)?

ilario, 2018-02-19:
The documentation is annoyingly quite spread, are you looking for something specific?

I link here the main information sources:

* on the website

https://libremesh.org/howitworks.html
https://libremesh.org/docs/en_quick_starting_guide.html
https://libremesh.org/docs/en_config.html
https://libremesh.org/docs/en_troubleshooting.html

* in the repositories

https://github.com/libremesh/lime-packages/blob/develop/README.md
https://github.com/libremesh/lime-sdk/blob/master/README.md
https://github.com/libremesh/lime-packages/blob/develop/packages/lime-docs/files/lime-example

* in university thesis

http://nicolasacco.diveni.re/~gio/internship-report/main.html
upcoming the one from @aparcar 

I think some local communities prepared some localized guides, maybe @patogit can link some?

bitchecker, 2018-02-20:
thanks for you links.

those are very useful, but, i don't understand if, with only simple router, i will be able to create a router of the mesh network that will be also an ap for users or not. it's a mix of ap/ad-hoc mode?

patogit, 2018-03-20:
> those are very useful, but, i don't understand if, with only simple router, i will be able to create a router of the mesh network that will be also an ap for users or not. it's a mix of ap/ad-hoc mode?
@bitchecker , there are many ways to set up a network. The difference between an access point and a router-to-router connection depends on what kind of antenna is used and on what kind of WiFi network is created. In general, if you set up two routers with LibreMesh close enough for them to connect with each other, they will connect automatically and also function as AP for users. This is most effective on dual-band routers (function in 2.4 GHz and 5 GHz simultaneously).

bitchecker, 2018-04-24:
> @bitchecker , there are many ways to set up a network. The difference between an access point and a router-to-router connection depends on what kind of antenna is used and on what kind of WiFi [...]

but in this case, 2.4Ghz is used for users and 5Ghz for routers by default or it depends from configuration?

i think that a real scenario implementation example will be help a lot on documentation.

ilario, 2018-10-01:
By default both (2.4 and 5 GHz) are used for both (AP and mesh).
One can specify that one interface has to be used just for mesh or just for AP as shown in the example [here](https://github.com/libremesh/lime-packages/blob/master/packages/lime-docs/files/lime-example#L82) (the example contains a typo: it's not ieee89211s but ieee80211s). It would be amazing if anyone could improve the documentation on this and fix the typo (good first PR).

ilario, 2018-10-01:
Alternatively one can include the [lime-smart-wifi](https://github.com/libremesh/lime-packages/blob/master/packages/lime-smart-wifi/) package, when compiling with [chef](https://chef.libremesh.org/) or downloading its package and installing via opkg (but I cannot find it...), that should set the 5 GHz for mesh and the 2.4 GHz for AP. Is this correct @aparcar?

closed 2018-12-08 by ilario

henmohr, 2019-11-06:
Some docs in PT-BR:
https://wiki.coolab.org/index.php/P%C3%A1gina_principal