libremesh/lime-packages #335: bmx7 iwinfo bandwidth estimation with autoconf wifi-iface

aparcar, 2018-02-20:
bmx7 offers a plugin to read the output of `iwinfo` and give a better estimation of the possible bandwidth. This doesn't work for now as LiMe configures the interfaces in a way that iwinfo/bmx7 doesn't detect them as wifi interface. We should think of a solution to offer more precise estimations of link quality. 

```
root@smpl-monitor:~# bmx7 -c interfaces
INTERFACES:
dev           state linkKey    linkKeys          type     channel rateMax idx localIp                     rts helloSqn rxBpP   txBpP   
wlan1-mesh_13 UP    DH2048M112 RSA896,DH2048M112 ethernet 0       54000K  2   fe80::6f0:21ff:fe33:db02/64 2   43463    382/2.5 196/1.3 
```
```
root@smpl-monitor:~# iwinfo 
wlan1-mesh ESSID: unknown
          Access Point: 00:00:00:00:00:00
          Mode: Mesh Point  Channel: 165 (5.825 GHz)
          Tx-Power: 17 dBm  Link Quality: 70/70
          Signal: -20 dBm  Noise: -95 dBm
          Bit Rate: 40.6 MBit/s
          Encryption: unknown
          Type: nl80211  HW Mode(s): 802.11abgn
          Hardware: 168C:002A 168C:3099 [Generic MAC80211]
          TX power offset: unknown
          Frequency offset: unknown
          Supports VAPs: yes  PHY name: phy1
```

@axn what do you suggest?

dangowrt, 2018-02-20:
I reckon you'll have to do something like here
https://github.com/freifunk-gluon/gluon/commit/4ce85afc22a2c64e5ac5ea1b52ad93f323e384d4
in the iwinfo plugin to first climb down to the 'physical'
interface.

axn, 2018-02-21:
Thx Daniel and Paul, I'll work that out...
/axel

aparcar, 2018-02-21:
https://github.com/bmx-routing/bmx7/pull/12

dangowrt, 2018-03-06:
@aparcar please test if https://github.com/bmx-routing/bmx7/pull/15 fixes the problem

aparcar, 2018-03-10:
now that bmx7 detects devices and speed correctly we should disable the max rate per default as the routing protocols decides it based on the actual situation. I'd suggest to disable the settings of rateMax if no specific rate is set.

@G10h4ck @p4u @dangowrt

closed 2018-03-11 by dangowrt