libremesh/lime-packages #345: anygw: alfred lease sharing is broken

dangowrt, 2018-03-11:
Doesn't work with dnsmasq on current develop branch. Maybe alfred is broken or doesn't work well on non-batman-adv bridge ports, I never really saw it working, so I have a hard time figuring out the expected behaviour...

nicopace, 2018-03-12:
@dangowrt will get to check this later hopefully... it is a thing that has had issues in the past, and also a very useful feature!

@aparcar bmx and batman-adv are part of a solution that work together.
Despite LibreMesh is flexible enough by design to support different kinds of setups, the collective decision has been for the default to be a collaboration between batman-adv (layer2 mesh routing) and bmx6 (layer 3 mesh routing).

Implementing the functionality of bmx7-sms can be done the same way there is bmx6-sms, and the same that there is for alfred... but as of now, there has not been a decision to change the default, that is the one @dangowrt is talking about.

aparcar, 2018-05-27:
related https://github.com/libremesh/lime-packages/issues/369

dangowrt, 2019-02-08:
Just to add my two cents to this:
 * lease sharing is only required for layer-2 mesh, so using bmx6/7-sms for this doesn't make sense to me, because for layer-2 mesh we use batman-adv (and potentially could use 802.11s with `mesh_fwding=1`), so alfred or something which works on top of either Ethernet layer-2 or IPv6 link-local multicast (like alfred) is the right thing here imho.
 * replacing our current hacky solution with ddhcpd could be nice, it looks like it is made for exactly this purpose. I didn't have a deep look at it nor try it myself, but it looks like it's DHCP (v4) only and no support for DHCPv6 what-so-ever (and we need DHCPv6 to work on prefixes longer then `/64`, because SLAAC/RA only works for at `/64` networks.
 * fixing alfred also looks like a good option because it will allow to use the same tool also for distributed name resolution and other purposes (and already supports v4 and v6 and it's rather easy to hook this into any DHCP/DHCPv6/DNS server daemon). So hence my question: Is this just about a bug or do you see a fundamental design flaw in our currentl solution or when using alfred+odhcpd in future?

p4u, 2019-02-08:
Agree with @dangowrt just add that alfred is not a good solution for DNS since it only works on the collision domain network. For DNS bmx-sms seems a much better option (actually is already implemented on bmx7-mdns package).

As an extra bonus flame topic, let me add that IMO layer2+layer3 libremesh firmware should be burned in hell. Just layer3 or just layer2 is simpler, works better and pobably fits 99% of use cases. 

Is there someone around the globe that is actually using the magic properties of mixing layer2+layer3 mesh routing? Whatever.

ilario, 2019-02-08:
> As an extra bonus flame topic, let me add that IMO layer2+layer3 libremesh firmware should be burned in hell.

Let's not flame here, I created a new issue for this: #468

nicopace, 2019-05-29:
@G10h4ck has implemented shared-state, that removes dependency with alfred so this issue is no longer relevant. closing.

closed 2019-05-29 by nicopace