libremesh/lime-packages #367: GroundRouting: connecting internet with static IP

nicopace, 2018-05-26:
USECASE: ISPs that give static ips instead of dhcp in the network.

Expected behaviour: the app to allow the user to set the ip.

G10h4ck, 2018-05-27:
I believe groundrouting is not related to this usecase

nicopace, 2018-05-30:
Many of the conversations we have been having about how to reimplement groundrouting using media detection (mainly with @nicoechaniz ) include detecting what happens on the other side of the wire for every wire that is connected.
that is, if there is a dhcp server, if there is another libremesh device, or if there is none, and how to proceed with each.

ilario, 2018-10-01:
So it would be something like an **automatic** configuration for ground routing?

nicopace, 2018-10-07:
that is what was talked... but no progress on this yet... any volunteer @ilario  ;)

ilario, 2018-12-08:
@nicopace reading again the original description of the issue, sounds more like #429, right?
Autoconfiguration of groundrouting sounds like something too epic to do (we usually refer with "groundrouting" to a very specific thing).

ilario, 2019-10-04:
This issue is not clearly defined, I'm closing. Please feel free to re-open explaining it better. If it was actually a duplicate of #429, please reopen that issue instead of this one.

closed 2019-10-04 by ilario