libremesh/lime-packages #369: Evaluate DDHCPD as alfread replacement

aparcar, 2018-05-27:
[DDHCPD](https://github.com/sargon/ddhcpd) could do some work for us automatically, couldn't it?

We should evaluate that and may do a migration do get rid of various existing problems.

nicopace, 2018-05-30:
@nicoechaniz

aparcar, 2018-07-02:
This could remove the need of dnsmasq-lease-share, I'll set up a test config and share it here

nicopace, 2018-07-02:
We need efficient mesh data distribution for many things (leases, dns, pitbull authorized macs, messaging, ...).
We need to abstract this in order to have it no matter what changes we do with the routing protocols.
@G10h4ck is this happening now?

aparcar, 2018-07-04:
So what do?

ilario, 2019-10-04:
[shared-state](https://github.com/libremesh/lime-packages/tree/master/packages/shared-state) has been implemented 7 months ago and replaces Alfred. Closing.

closed 2019-10-04 by ilario