libremesh/lime-packages #375: anygw: replace dnsmasq-dhcpv6 with odhcpd-ipv6only

aparcar, 2018-06-04:
The latter is already standard for OpenWrt and comes preinstalled. It still creates entries in `/tmp/hosts` so DNS (sharing) shoudl continue to work.

aparcar, 2018-06-13:
related https://github.com/openwrt/odhcpd/issues/113

nicopace, 2018-06-20:
It is definitely one of the pain in the ass to be using dnsmasq as it breaks all the time.
I don't know how odhcpd behaves, we should test it out for robustness? or should we trust blindly? :)

aparcar, 2018-06-20:
odhcpd-ipv6only is used in LEDE stable and the upcoming 18.06 release, so it should be fairly stable. If I remember correct @dangowrt suggested this before, so I'm just following up on that.

odhcpd as a full dnsmasq dhcp replacement brings the current limitation that odhcpd does not (yet https://github.com/openwrt/odhcpd/issues/113) allow any lease input, meaning lease sharing wouldn't work. However, thats's may be tackeld by https://github.com/libremesh/lime-packages/issues/369

aparcar, 2018-12-09:
@dangowrt can you have a look at this and the ubus request posted a long time ago? https://github.com/openwrt/odhcpd/issues/113

ilario, 2019-10-04:
Seems that we gave up on this, as after years we still use dnsmasq.
Good thing is that dnsmasq seems more stable now.
I'm closing, please re-open if I'm wrong and this is still an open issue.

closed 2019-10-04 by ilario

ilario, 2019-10-04:
Just realized that this was a duplicate of #294 and was being addressed by @aparcar in #379