libremesh/lime-packages #383: broken client mode for wireless

aparcar, 2018-06-17:
```
config wifi 'radio0'
	list modes 'client'
	option client_key 'pass'
	option client_ssid 'ssid'
	option client_encryption 'psk2'

config net 'wirelessclient'
	option linux_name 'wlan0-client'
	list protocols 'wan'
```

creates the following output in `/etc/config/wireless`
```

config wifi-iface 'lm_wlan0_sta_radio0'
	option mode 'sta'
	option device 'radio0'
	option ifname 'wlan0-sta'
	option network 'lm_net_wlan0_sta'
	option ssid 'ssid'
	option encryption 'psk2'
	option key 'pass'
```

However I think `lm-net_wlan0_sta` isn't created. The official script simplify creates an `wwan` interface with dhcp and wan firewall enabled, I think we should do the same.

ilario, 2018-10-01:
Check #127, I think it's related.

nicopace, 2018-10-07:
@ilario how would you recommend to proceed?
@aparcar what would be the expected input?

ilario, 2018-10-07:
I don't know if it's a possible solution, but I just tried to include both client interface (wlan0-sta) in bat0 on the client and AP interface (wlan0-ap) in bat0 on the access point and it worked: the nodes could see each other.
So, instead of including it in br-lan (not possible, see #127), I included it in bat0.

ilario, 2018-10-22:
Sorry, I misunderstood the problem, it is a mere error in the name of the interface, fixed the example in #414, I move the discussion on client + lan here: #415

nicopace, 2018-10-22:
can we close this issue then @ilario @aparcar ?

PR #414 (2018-10-25): lime-example improved documentation for wireless client mode
The documentation of wireless client mode was a bit incorrect: the wireless interface is not wlan0-client but wlan0-sta, this should fix #383 
This also adds a comment recommending to check if the used radio name exists
closed 2018-10-25 by ilario