libremesh/lime-packages #389: create easy, reliable virtual machine workflow for testing

patogit, 2018-06-21:
Currently there are three sets of instructions to run LibreMesh in a virtual machine, and none of them work for me. I use Manjaro (Arch derivative). Right now I need this in order to test language handling and translation of LuCi and LimeApp, so any option that doesn't let me access those things isn't useful.

1. in the lime-sdk Readme https://github.com/libremesh/lime-sdk/blob/master/README.md#using-plain-qemu. This part is incomplete: "Using plain QEMU. Plain qemu can be launched straight from the command line, if you don't need to access LibreMesh web interface and just want to have a shell you can issue [...] You can also have access to the web interface configuring a tap device on the host as follows. TODO". Then there's the part in "Using Virt-Manager" that says "Now you have to create a new virtual machine: click on File/New Virtual Machine, select Import existing disk image and choose the arm architecture under Architecture options and virt machine type." But when I launch Virt-Manager, I don't see the option "Import existing disk image".
2. by @gmarcos87 https://github.com/gmarcos87/lime-virtualbox These instructions get me all the way to a VM mesh node that's visible to other nodes on my network, but its web interfaces won't load (LuCi and LimeApp) when I go to the VM's IP in my browser. And sometimes it doesn't finish booting, so I restart until it finishes booting.
3. by @nicopace https://blog.freifunk.net/2017/07/22/streamline-app-development-openwrt/ this doesn't work, gives an error `qemu-system-x86_64: -enable-kvm: unsupported machine type. Use -machine help to list supported machines`.

Can we figure out something that works more reliably? At least on Debian and Arch-based distros? And preferably on Mac and Windows too, since some people will probably find it useful someday. I don't know much about the options, but maybe Docker, Vagrant or LXC are options? I already use Docker to run `cooker`. LXC was promised to be the future of lighweight virtualization years ago, but I haven't seen it in use.

aparcar, 2018-06-23:
would it make sense to offer Docker images as?

aparcar, 2018-06-23:
Hi @patogit , I started to work on automatic Docker builds [here](https://github.com/aparcar/libremesh-docker) and could move it to the official libremesh account once approved. 

Pleaes give the Docker image `aparcar/libremesh` a try, the tags `stable` (17.06) and `latest` (openwrt snapshot + lime-packages master branch) are available.

I have no experience with LXC/qemu on Mac/Windows so I'd suggest we create a CI which builds both, rootfs filesystems for docker images.

patogit, 2018-06-23:
Hi @aparcar , great that you've made this, and I don't know what to do with it. The only Docker command I know is the one to cook:
```
sudo systemctl start docker
cd ~/Apps/lime-sdk/lime-sdk
sudo docker run -v "$(pwd)":/app cooker -c x86/generic --profile=Generic --flavor=lime_default
```
1) What command(s) do I run in order to test the images you created?
2) How do I build them with modified files, (for example I modify a translation file or a language selection menu)?
3) Does it matter at all that the Docker images are x86_64 and the routers are other architectures?

aparcar, 2018-06-23:
1. I'll create a README.md
2. as for 1.
3. No that should not matter for your translation use case

patogit, 2018-06-23:
> 1. I'll create a README.md

Great!
> 3. No that should not matter for your translation use case

The most complicated thing I'm modifying is https://github.com/libremesh/lime-packages/issues/360 but even this is pretty basic stuff where the architecture doesn't matter as far as I know, all LuCi interface stuff.

aparcar, 2018-06-23:
check out the [README.md](https://github.com/aparcar/libremesh-docker/blob/master/README.md)

patogit, 2018-06-23:
Seems like we're moving a good direction, as this is all very easy and lightweight (no huge downloads or long build times), but it's not working yet:
```
[user@system libremesh-docker]$ sudo docker run -it aparcar/libremesh:latest sh
Unable to find image 'aparcar/libremesh:latest' locally
latest: Pulling from aparcar/libremesh
151046cb364c: Pull complete 
bddae430b96a: Pull complete 
e208ed797d54: Pull complete 
02202d7be44b: Pull complete 
Digest: sha256:a4dadc7c8e3e0580d71e2ddc635cc1081779cc4ebee605a6363e1ce7e0264800
Status: Downloaded newer image for aparcar/libremesh:latest

BusyBox v1.28.4 () built-in shell (ash)

/ # ip r
default via 172.17.0.1 dev eth0 
172.17.0.0/16 dev eth0 proto kernel scope link src 172.17.0.2 
/ # 
```
Then in my browser I go to those two IP addresses and nothing... makes sense, as those IPs aren't in the subnet of my LibreMesh network. Maybe because it's the latest version, and different versions of LibreMesh don't tend to be compatible.

So I try stable:
```
[user@system libremesh-docker]$ sudo docker run -it aparcar/libremesh:stable sh
Unable to find image 'aparcar/libremesh:stable' locally
docker: Error response from daemon: manifest for aparcar/libremesh:stable not found.
See 'docker run --help'.
[user@system libremesh-docker]$ 
```

patogit, 2018-06-23:
> Then in my browser I go to those two IP addresses and nothing... makes sense, as those IPs aren't in the subnet of my LibreMesh network. Maybe because it's the latest version, and different versions of LibreMesh don't tend to be compatible.

Yesterday with a VirtualBox image following @gmarcos87 instructions, the new LibreMesh node had a local IP and appeared on the list of nodes when I went to thisnode.info on my router.

aparcar, 2018-06-23:
@patogit thanks for testing, I'll look into this. Please post all further issues on the repo aparcar/libremesh-docker

patogit, 2019-02-16:
looks like @gmarcos87 has made some progress on this in https://github.com/gmarcos87/lime-virtualbox

aparcar, 2019-02-16:
@patogit 2 years old?

patogit, 2019-02-16:
Hmmmmm. My mistake. @gmarcos87 posted about it in a recent thread and I mistook it for something new, without looking at the last commit date.

aparcar, 2019-02-16:
maybe we can create an image with the ci.

nicopace, 2019-05-29:
I was able to make it work with qemu and networking:

```bash
$ ip tuntap add name tap0 mode tap
$ ifconfig tap0 up
$ ip addr add 192.168.254.1/24 dev tap0

$ qemu-system-x86_64 \
-m 128 \
-smp 1,sockets=1,cores=1,threads=1 \
-no-user-config \
-nographic \
-nodefaults \
-drive file=openwrt-18.06.2-x86-64-combined-squashfs.img,format=raw,if=none,id=drive-ide0-0-1 \
-device ide-hd,bus=ide.0,unit=1,drive=drive-ide0-0-1,id=ide0-0-1,bootindex=1 \
-serial mon:stdio \
-netdev tap,id=hostnet0,ifname=tap0,script=no,downscript=no \
-device e1000,netdev=hostnet0,id=net0,mac=52:54:00:86:34:95,bus=pci.0,addr=0x3 

# inside the vm
$ ip addr add 192.168.254.2/24 dev eth0

# outside the vm this will open a browser with luci (now)
$ xdg-open http://192.168.254.2/ 
```

hope it helps

PR #702 (2020-04-30): Qemu development tools
This PR adds two scripts to ease running qemu images. 
Am option `--libremesh-workdir` is provided to dynamicaly modify the rootfs with a local libremesh repo so that qemu boots directly with the development files. This is very handy to test boot scripts, etc. It is not perfect as it does not delete the old files already present in the images. To support this I modified all the packages to follow the "files directory scheme".

I have built and uploaded images to repo.libremesh.org that will work with this scheme.

I still did not investigate rootfs persistance, so it is  [... cut]
closed 2020-04-30 by spiccinini