libremesh/lime-packages #390: mesh-dns: design uci config

aparcar, 2018-06-23:
How should the config file of `mesh-dns` look like? Here a possible approach

```
config main "main"
    list trusted_tld "mesh" # other tlds are ignored
    list trusted_tld "qmp"
    option share_hosts_file "true" # to share /etc/hosts

config alfred "alfred"
    option id 70 # may use other number if this is already taken

config bmx7 "bmx7"
    option nodes "true" # add bmx7 short ID and hostname as DNS entry using crypto IPv6
    list trusted_id "all" # if not all, list of IDs to trust as they're not in the local mesh

config dns "chat_mesh"
    option name "chat.mesh"
    option ipv4 "1.2.3.4"
    option ipv6 "2002:d161:ee7f:1::9b9"
```

nicopace, 2018-06-23:
I would say let's not get this out of /etc/config/lime-defaults cause it is what we are using now for FirstBootWizard.
Or... we will need to create a bridge from /etc/config/lime-defauts to mesh-dns

aparcar, 2018-06-25:
I see this as two different things. Implementing a script that uses entries from lime-defaults as dns entries is fairly trivial (aka copying it to /hosts/tmp/lime-defaults-dns && dnsmasq reload) while the (my) idea of mesh-dns is that hosts setup entries and share them over the network.

Combining both results in the problem of entry changes, you'd have to update all previously setup nodes else it'd come to doubled DNS entries.

ilario, 2019-10-04:
Closing, see comments on #303 (shortly, if we don't ship BMX* by default, this issue has a very low priority).

closed 2019-10-04 by ilario