libremesh/lime-packages #403: Include an option to spoof internet connectivity

patogit, 2018-09-14:
In order to avoid the automatic disconnection of phones from LibreMesh WiFi networks when the network does not have internet connectivity, it is necessary to fool the phone by spoofing internet connectivity. Some networks may not want this behavior, so it should be an option that is deactivated by default.

For example, my Android 6.0.1 phone automatically disconnects from any WiFi network within 30 seconds of detecting that there is not internet connectivity, and there is no option to disable this behavior in the phone. This means that I cannot use the phone on the local WiFi network for file transfers, configuring routers and antennas or other local network activities.

Some links to start understanding:

- https://android.stackexchange.com/questions/123129/how-does-wifi-in-android-detect-if-the-device-has-to-sign-in-or-not
- https://socifi-doc.atlassian.net/wiki/spaces/SC/pages/94371841/DNS+Workaround+to+keep+Android+Splash+Page+and+the+Captive+Portal+Notification+active

patogit, 2018-09-14:
continuing the thread started here https://lists.libremesh.org/pipermail/lime-users/2018-August/001275.html

ilario, 2018-11-12:
This issue is much related to #93. For avoiding such instabilities, a couple of hysteresis parameters should be added to Watchping.
Right now with just one failed ping or one successful ping, Watchping changes the BMX6 configuration. With hysteresis X failed pings would need to fail in order to instruct BMX6 to stop sharing and Y successful pings would be needed in order to re-activate it.

gmarcos87, 2018-11-14:
Hello @patogit, we came across this situation testing "pirania" (captive portal that ate a pitbull in the Amazon). The solution proposed by both android and ios is very bad. Even worse, Wathsapp detected the captive portal and insisted that I forget the network and delete it from my list of accepted networks. It is a very painful situation and we can do little with the ridiculous decisions of these corporations.

ilario, 2018-12-08:
@gmarcos87 so you're saying that we should not do this?
I think that adding histeresis (explained above) would solve without causing the "forget the network" problem.

ilario, 2018-12-08:
I just realized this is a duplicate of #93

ilario, 2019-10-04:
From the discussion, looks like we should close this issue as the mobile phones does not like at all fake announcements of internet connectivity.

closed 2019-10-04 by ilario