libremesh/lime-packages #408: Static proto: specify gateway

ilario, 2018-10-07:
When a static IPv4 address is set via specific interface configuration, there should be a way to set also the gateway, needed when configuring it for a WAN.

In /docs/lime-example one can read:

```
list protocols 'static:172.23.0.1/24'		# Set up a static IP (both IPv4 and IPv6 supported, specify twice one per IP type)
```
The gateway could be accepted as a second optional argument, like this:

```
list protocols 'static:192.168.1.2/24:192.168.1.1'
```

nicopace, 2018-10-07:
This is implemented in here: 
https://github.com/libremesh/lime-packages/blob/d2ba7a7b85ac6f4bb26741fd365016dccb39545d/packages/lime-system/files/usr/lib/lua/lime/proto/static.lua#L16

anyone who might want to implement what @ilario is proposing should start from there.

I am not sure if it is fine to define that there... @G10h4ck ?

ilario, 2018-11-10:
Closing this and continuing on #409: from the discussion there seems that the issue is basically the same.

closed 2018-11-10 by ilario