libremesh/lime-packages #409: Static proto: document how to set an IPv6

ilario, 2018-10-07:
Static proto does support IPv6 but I didn't get how to do that...
In lime-example it says:
```
list protocols 'static:172.23.0.1/24' # Set up a static IP (both IPv4 and IPv6 supported, specify twice one per IP type)
```
but setting IPv6 fails due to the colon (:) being interpreted as arguments separator instead of part of the IPv6, escaping them with backslash does not work neither.

nicopace, 2018-10-07:
@G10h4ck can you jump on this?

G10h4ck, 2018-10-07:
This comment on the code https://github.com/libremesh/lime-packages/blob/master/packages/lime-system/files/usr/lib/lua/lime/proto/static.lua#L21 suggests it is supported, I don't know if Pau tested it,  @ilario can you give more information about the error you get?
Anyway the code seems a bit strange to me and I would be surprised if it worked... moreover it miss how to set a gateway and other things usually needed in static setups

nicopace, 2018-10-07:
What is strange about it?

how would you use it?

Can you give us an example?
Like, one per option:
* ipv4
* ipv6
* ipv4+ipv6

thanks!

ilario, 2018-10-07:
> moreover it miss how to set a gateway and other things usually needed in static setups

Exactly, see #408 

> can you give more information about the error you get?

The colon is interpreted as an arguments separator instead that being interpreted as part of an IPv6.
For example, if I set:
```
list protocols 'static:192.168.1.5/24'
list protocols 'static:aaaa::1111/64'
```
I obtain the following error:
```
# lime-config
Clearing wireless config...
Clearing network config...
Disabling odhcpd
Cleaning dnsmasq
Disabling 6relayd...
/usr/lib/lua/lime/network.lua:276: attempt to index local 'addr' (a nil value)
stack traceback:
	/usr/lib/lua/lime/network.lua:252: in function </usr/lib/lua/lime/network.lua:252>
	/usr/lib/lua/lime/network.lua:276: in function 'createStaticIface'
	/usr/lib/lua/lime/proto/static.lua:23: in function 'setup_interface'
	/usr/lib/lua/lime/network.lua:251: in function </usr/lib/lua/lime/network.lua:251>
	[C]: in function 'xpcall'
	/usr/lib/lua/lime/network.lua:251: in function </usr/lib/lua/lime/network.lua:210>
	[C]: in function 'xpcall'
	/usr/bin/lime-config:45: in function 'main'
	/usr/bin/lime-config:65: in main chunk
	[C]: ?
Configuring system...
Let uhttpd listen on IPv4/IPv6
```
but looking at the code (quite weird one, as @G10h4ck pointed out) I realized that something like this:
```
list protocols 'static:192.168.1.5/24'  
list protocols 'static:aaaa:0000:0000:0000:0000:0000:0000:1111/64'
```
works. So it works just if all the 8 fields are specified, as they get recognized as 8 arguments and reassembled in the "workaround".

> Can you give us an example?

Meh, I think that the choice of a colon : as arguments separator was a bad choice and should be changed [here](https://github.com/libremesh/lime-packages/blob/f6b0f9a3ed316c37f548d891b6051faca2827246/packages/lime-system/files/usr/lib/lua/lime/network.lua#L13) but this will break many network-profiles.

An alternative solution is to improve the currently used parser, it will still be weird code but it can be made so that it can accept shortened IPv6 addresses.

aparcar, 2018-10-08:
What would be a better separator? I guess we could change it and add a fallback. All ipv6 examples should show the new separator

G10h4ck, 2018-10-09:
no separator at all, i remember to implement options for protocols with another syntax `option yourprotoname_youroptionname yourbelovedvalues` you can see it in action here https://github.com/libremesh/lime-packages/blob/master/packages/lime-docs/files/lime-example#L45

so I suggest to just take a look at how it is implemented for other protocols and do that for `proto static` too, and bleach that piece of clumsy code once and for all...

ilario, 2018-10-26:
@G10h4ck sounds like a good idea, I had a look at it, currently the values of the configurations you're referring to are obtained via `config.get("network", "option_name")` or `config.get("wifi", "option_name")` or `config.get("system", "option_name")` which looks into `lime` section in the /etc/config/lime* files.

Now we're referring to a config that would be in one of the possibly many `net` sections, how do I get the subsection name that caused the call to `static.lua`?
In other words, I understand that the value could be obtained with the more generic `config.uci("net", "subsection", "option_name")` but how do I get the "subsection" value?

G10h4ck, 2018-10-27:
There is no such thing as "subsection".
"network", "wifi" and "system" are section names, and the three of them have type "lime".
In case of specific sections "net", "wifi", "hwd_gr" and "bgp_peer" is the type of the section, while the name is arbitrary (it may be even an anonymous section (without name) but we suggest to specify a name when possible), and as a consequence you cannot rely on the section name to access to the configurations. You could eventually iterate over all the section of certain type to look for the one you need, but that would be clumsy and doesn't seems a reliably solution, so I have made this PR https://github.com/libremesh/lime-packages/pull/420 so the specific section data is passed down and you can access it via `args._specific_section` in your `lime.proto` methods if used by a specific section once it the PR is merged

ilario, 2018-11-10:
Would it be enough to implement the following options

- static_ipv4
- static_gateway_ipv4
- static_ipv6
- static_gateway_ipv6

or a way to set arbitrary options is needed?

For [static IPv4](https://openwrt.org/docs/guide-user/network/ipv4/start#protocol_static_ipv4) there's not much more needed (DNS? is going to be useful a setting in the specific interface configuration if resolvers are already defined in the general one?).

For [static IPv6](https://openwrt.org/docs/guide-user/network/ipv6/start#protocol_static_with_ipv6) there's more stuff, are some of these a must-have?

ilario, 2018-12-08:
Can anyone review #427?
It has some limitations (e.g. no multiple IPv4 or IPv6 can be set on one interface, but quite nobody uses this anyway) but should solve.

PR #427 (2018-12-19): Static proto: add setting for gateway and clean IPv6 setting
Depends on #420 
This should fix #409 and #408 
Changes the settings of a static IP from
```
	list protocols 'static:1.2.3.4/24'
	list protocols 'static:1:2:3:4:5:6:7:8/64'
```
to
```
	list protocols 'static'
	option static_ipv4 '192.168.1.2/24'
	option static_gateway_ipv4 '192.168.1.1'
	option static_ipv6 '2a00:1508:0a00::1234/64'
	option static_gateway_ipv6 'fe80::1'
```
the advantages are: possibility to set a gateway, support for any IPv6 format (compressed with :: or not).
closed 2018-12-19 by G10h4ck