libremesh/lime-packages #415: Wifi client mode + LAN

ilario, 2018-10-22:
Currently the wifi client mode works just if used as WAN, so with a natting and a DHCP client for having the internet access.
To build a point to multipoint (as opposed to mesh) network can be a great option in many cases (for example when the topology is a tree or when the hardware does not support neither adhoc nor ieee80211s or if we want to use the better supported ap-sta mode rather than mesh) and being able to use wifi client in LAN network is what's missing.
The main trouble in doing this is described [here](https://wiki.openwrt.org/doc/howto/clientmode#bridged_client_mode_issues).
A proposed solution is to include the wlan0-sta interface in bat0 bridge instead than in br-lan bridge.
Related issues: #127, #383, #47

nicopace, 2018-10-22:
Hi ilario,
I never came across this setup... can you share a more detailed example on how would you expect to use it, and if you have a rough idea on how to implement it?

ilario, 2018-10-26:
I will copy here a comment I made originally elsewhere, tell me if it is clear to you:

> a user could want to build a point to multipoint network with hardware that does not support adhoc nor ieee802.11s (I was planning to do this with a lot of modem-routers we received as donations with the local hacklab), for doing this some nodes should be configured as client via radio specific configuration or general configuration (unsupported, I think), see #415 for some problems hindering this

Another reason for using AP-STA as a backbone could be a better support in the drivers for these two modes rather than for adhoc or for ieee802.11s, so for performances and stability reasons.

If the inclusion in bat0 is a solution (it worked for me, maybe there's a better one? @G10h4ck?) I would suggest to expand [this line in lan.lua](https://github.com/libremesh/lime-packages/blob/master/packages/lime-system/files/usr/lib/lua/lime/proto/lan.lua#L46) into a section managing the interface inclusion in bat0 (is there also the risk that this interface configuration happens before bat0 is up?)

nicopace, 2018-11-03:
Awesome.
No particular interest of having this kind of networks, neither know anyone that uses them (other than traditional WISPs), but please, no problem on supporting it, sound like falls into something we could be interested in having as a collective.

G10h4ck, 2018-11-03:
wifi client can't be used in LAN because there is no standard way to bridge a wifi STA interface

ilario, 2018-11-04:
@G10h4ck read above

ilario, 2018-11-10:
@nicopace for example, in NinuxRoma they use [AP-sta with original firmware](http://wiki.ninux.org/HowToConfigurazioneAirOSStepByStep_06) for [maximizing link performances](https://diyisp.org/dokuwiki/lib/exe/fetch.php?media=technical:ninuxroma-routingarchitecture-documentversion1.pdf) and they manage the routing via [ground routing approach](https://docs.google.com/document/d/1tEp4rFNDis4RAwwo4wwg6xDMdqeJKAi8FVYh_FsPEdk)

ilario, 2018-11-10:
@G10h4ck suggested to specify batadv as a protocol for this specific interface configuration, something like:

```
config wifi 'radio0'
	list modes 'client'
	option channel_2ghz 'auto'
	option client_ssid 'someRandomAP'
	option client_key 'thepassword'
	option client_encryption 'psk'

config net 'wirelessclient'
	option linux_name 'wlan0-sta'
	list protocols 'batadv:%N1'
```

instead of what I though that had the last line with `list protocols 'lan'`.

The problem of this is that it does not setup the sta interface, so the resulting /etc/config/network is broken.
We need a proto/client.lua (copying the proto/ieee80211s and replacing all ieee80211s occurrences with client works) and then, important, the specific configuration has to include a list protocols 'client' in order to activate proto/client.lua

patogit, 2018-11-29:
I think this functionality is very important. Being able to connect to the internet gateway via WiFi instead of cable is very useful and can make network deployment less expensive, faster, and more flexible. See this thread for an example https://lists.libremesh.org/pipermail/lime-users/2018-November/001339.html

ilario, 2018-12-03:
@patogit I think you refer to something that is already implemented in mode client [here](https://github.com/libremesh/lime-packages/blob/master/packages/lime-system/files/usr/lib/lua/lime/mode/client.lua) and documented [in lime-example](https://github.com/libremesh/lime-packages/blob/bd968220bff199245f227a8b7e3aa017fb307df7/packages/lime-docs/files/lime-example#L105-L119). See [my answer to the email thread](https://lists.libremesh.org/pipermail/lime-users/2018-December/001348.html). This issue is about more or less the same thing but for connecting two LibreMesh routers via AP+client, instead you're looking to connect a non-LibreMesh gateway to a LibreMesh router via AP+client.

ilario, 2019-01-27:
#426 has been merged and enables the usage of BMX6 on the AP-client links.
Until when some non-BMX6 networks (e.g. NinuxRoma with only OLSR on WDS AP-client links or eigenNet with only BATMANadv on adhoc) ask for this feature to be expanded to their scenario, I would consider this issue closed :)

ilario, 2019-06-24:
Seems that [BMX6 is not going to be included in next release](https://github.com/LibreRouterOrg/openwrt/commit/46525b36892694d118447a42e945b4df7ad09eae) while [Babeld will be](https://github.com/LibreRouterOrg/openwrt/commit/d9be9446aa572efeb2514a30b29685aa9a3cf1d7).

So the current solution in #426 has to be adapted to work with Babeld.
Both [BMX6](https://github.com/libremesh/lime-packages/blob/master/packages/lime-proto-bmx6/src/bmx6.lua#L171) and [Babeld](https://github.com/libremesh/lime-packages/blob/master/packages/lime-proto-babeld/src/babeld.lua#L95-L96) are disabled on AP interfaces but BMX6 was running on it anyway as the AP interface was into br-lan which is included in BMX6.
I hope that the same situation happens also for Babeld, so that the adaptation of #426 will be straightforward.

commit 69ed489f (2019-09-08): Merge pull request #554 from ilario/apsta_backbone

Allow routing on top of AP-client links, rewrite of #551, fix #415
closed 2019-09-08 by ilario