libremesh/lime-packages #423: Proposal: Simple reporting command for helping debug

ilario, 2018-11-07:
Just an idea: it could be useful to have a command for helping bug reporting.
This would get debugging info from various sources and output it in an agglomerated form, optionally to a file.
It could be included in the `lime-debug` packages.

A list of data that could be included:

- /etc/config/lime (removing passwords)
- /etc/config/lime-defaults
- /etc/config/network
- /etc/config/wireless (removing passwords)
- dmesg
- batctl if
- batctl o
- bmx6 -c show=status show=interfaces show=links show=originators show=tunnels
- free
- ps
- ip address show
- ip route show
- brctl show
- ip link show
- /proc/cpuinfo
- df
- ...?

Additional interesting data can be:
list of ieee802.11s connected peers (how can one obtain this?)
LibreMesh and OpenWrt version
...?

aparcar, 2018-11-07:
Not bad, should I implement that?

ilario, 2018-11-07:
Up to you :)
I would make the command in a way that without arguments prints everything to STDOUT (so that it can be piped to sprunge or copied and pasted) and if an argument is given everything is saved in a file with that name.

ilario, 2018-11-08:
This is an example of a situation where even with all the aforementioned data I cannot understand why neither BATMAN adv nor BMX6 can see the peers over established IEEE802.11s links:

https://lists.libremesh.org/pipermail/lime-users/2018-November/001331.html

Which piece of information are we missing?

aparcar, 2019-01-23:
We could setup something like debug.libremesh.org using https://github.com/solusipse/fiche
The rest is quite easy, I'll setup a package within the next days

aparcar, 2019-01-23:
https://github.com/libremesh/lime-packages/pull/457/files

PR #556 (2019-10-07): lime-report: status reporting utility, adaptation of #457, fixes #423
Here I modified @aparcar's lime-report from #457.
I also added it as a dependency of lime-debug.
As a future modification, we can adapt lime-report to use liblognorm rules if installed, see [here](https://github.com/libremesh/lime-log-review).

Please tell me if it is ok to have hardcoded tests or if they should be taken via UCI from a `/etc/config/lime-report` file.

Fix #423
commit 485cf114 (2019-10-07): Merge pull request #556 from ilario/lime-report_resurrecturis

lime-report: add status reporting utility, adaptation of #457, fixes #423
closed 2019-10-07 by spiccinini