libremesh/lime-packages #424: lime-librenet6 package missing title?

ilario, 2018-11-08:
A user reported that compilation with lime-sdk throws an error about lime-librenet6:
```
Makefile:33: *** Package/lime-librenet6 is missing the TITLE field.  Stop.
```
I don-t understand, as in the Makefile the field is defined, maybe the `/Default` [here](https://github.com/libremesh/lime-packages/blob/17.06/packages/lime-librenet6/Makefile#L11) could cause this? No idea.

On a side note: why lime-librenet6 is present in 17.06 branch but not in master?

ilario, 2019-10-04:
@aparcar @G10h4ck why isn't this package also in master branch? Has it been obsoleted by something else?

G10h4ck, 2019-10-04:
I requested modification in the PR, but they where never made. It is in the other branch because @nicopace merged it by error and we never reverted it

ilario, 2019-10-04:
Got it, you refer to the merge of #400. 
Your comments are in an open issue on #579

ilario, 2023-03-03:
Is librenet6 still functional? Anyway the package is not in the master branch. Closing.

closed 2023-03-03 by ilario