libremesh/lime-packages #430: In lime-defaults `ap_ssid` required even when `list_modes 'ap'` is removed

Juul, 2018-11-20:
In `config lime 'wifi'` even if we remove `list modes 'ap'` it still seems to required `ap_ssid`.

Error:

```
WARNING: Attempt to access undeclared default for: wifi.ap_ssid
stack traceback:
/usr/lib/lua/lime/config.lua:24: in function 'get'
/usr/lib/lua/lime/utils.lua:61: in function 'network_id'
/usr/lib/lua/lime/utils.lua:81: in function 'applyNetTemplate10'
/usr/lib/lua/lime/network.lua:55: in function 'primary_address'
/usr/lib/lua/lime/proto/lan.lua:16: in function 'configure'
/usr/lib/lua/lime/network.lua:230: in function </usr/lib/lua/lime/network.lua:230>
[C]: in function 'xpcall'
/usr/lib/lua/lime/network.lua:230: in function </usr/lib/lua/lime/network.lua:210>
[C]: in function 'xpcall'
/usr/bin/lime-config:45: in function 'main'
/usr/bin/lime-config:65: in main chunk
[C]: ?
/usr/lib/lua/lime/utils.lua:63: attempt to concatenate local 'network_essid' (a nil value)
stack traceback:
/usr/lib/lua/lime/network.lua:231: in function </usr/lib/lua/lime/network.lua:231>
/usr/lib/lua/lime/utils.lua:63: in function 'network_id'
/usr/lib/lua/lime/utils.lua:81: in function 'applyNetTemplate10'
/usr/lib/lua/lime/network.lua:55: in function 'primary_address'
/usr/lib/lua/lime/proto/lan.lua:16: in function 'configure'
/usr/lib/lua/lime/network.lua:230: in function </usr/lib/lua/lime/network.lua:230>
[C]: in function 'xpcall'
/usr/lib/lua/lime/network.lua:230: in function </usr/lib/lua/lime/network.lua:210>
[C]: in function 'xpcall'
/usr/bin/lime-config:45: in function 'main'
/usr/bin/lime-config:65: in main chunk
[C]: ?
```

[Example of non-working config file](https://github.com/libremesh/network-profiles/blob/master/peoplesopen.net/common/etc/config/lime-defaults#L25)

Juul, 2018-11-20:
Tagging @nicopace

nicopace, 2018-11-28:
I would propose to change the name of ap_ssid to network_name... 
That makes sense cause you also need that for libremap, and we could use it for ssid, hashes, etc... and we are not requiring a parameter that should be optional

ilario, 2019-08-02:
In #554 I added a comment on this in lime-example, we could consider this fixed after that.

ilario, 2019-10-04:
> I would propose to change the name of ap_ssid to network_name...

> +1 and the ESSID could be equal to the network_name by default but also different, without consequences on BATMAN adv cloud

I think these comments are still valid.
This change should be reflected in the lime-app interface also.
Maybe it's not worth it and we should just leave things like they are now?
Opinions? @nicopace @spiccinini @G10h4ck @gmarcos87

gmarcos87, 2019-10-04:
Ok, that's one option the other is to have an ap_ssid as fallback in lime-defaults or lime-default-factory

ilario, 2023-03-03:
This is a minor issue, it would be great to fix this but in 4 years nothing happened and things works anyway.
Feel free to re-open if you want to fix this.

closed 2023-03-03 by ilario