libremesh/lime-packages #433: Remove luci dependency

nicopace, 2018-11-28:
This package shouldn't depend on luci:
https://github.com/libremesh/lime-packages/blob/d2ba7a7b85ac6f4bb26741fd365016dccb39545d/packages/lime-map-agent/Makefile#L24

ilario, 2019-10-04:
@nicopace do you mean all the luci* dependencies?

ilario, 2019-10-04:
Maybe I got what you mean (investigating on #609): `luci-basic` is a dependency of `luci-lib-libremap` which is selected by `libremap-agent` which is selected by `lime-map-agent`.
Which point of this chain has to be broken?

ilario, 2019-11-21:
TL;DR: current implementation of libremap agent simply needs LuCI.

I thought that for breaking this dependency chain which causes #609 we could remove `luci-base` dependency in `luci-lib-libremap` (spoiler: no way)

in order to do so, we need to remove the usage of things from `luci-base`: 

* `luci.sys` gets used in `luci-lib-libremap`'s libremap.lua for few things:
  - sys.exec is a redirect to luci.utils library (also included in luci-base):
https://github.com/openwrt/luci/blob/1b835a50aabfbfe7d8d16a6d15c853315dc7ba79/modules/luci-base/luasrc/sys.lua#L25
which anyway just executes the command with io.popen:
https://github.com/openwrt/luci/blob/f73da9ca82ff8e0352c3942ea78ab7d1d0317668/modules/luci-base/luasrc/util.lua#L597-L603
  - sys.hostname just gets the hostname, can be done directly with nixio:
https://github.com/openwrt/luci/blob/1b835a50aabfbfe7d8d16a6d15c853315dc7ba79/modules/luci-base/luasrc/sys.lua#L36-L37
* `luci.httpclient` also depends on luci.util provided by luci-base and many other luci-base stuff

So I would say there's no way to disentangle luci-lib-libremap from luci-base.

Checking out [libremap-agent executable](https://github.com/libremap/libremap-agent-openwrt/blob/master/libremap-agent/files/usr/sbin/libremap-agent) turns out that it is also strictly bound to luci libraries and to luci-lib-libremap.

So seems that this issue cannot be solved.
If we want the libremap agent (do we?), we will have all this dependency cascade including the LuCI bloat.

spiccinini, 2019-11-21:
Great work @ilario documenting the dependencies!

closed 2019-12-03 by ilario