libremesh/lime-packages #438: lime-docs doesn't build

dangowrt, 2018-12-16:
lime-docs fails to build due to wrong source URL (`svn` proto on a github subdirectory -- that cannot ever have worked)

ilario, 2018-12-17:
It used to work and Github [officially supports Subversion](https://help.github.com/articles/support-for-subversion-clients/). Now seems that with Subversion 1.11 client there's a new and more strict check for server answer; and the one implemented by [Github does not pass the check](https://lists.apache.org/thread.html/90c028421ac86f787c2101ab61a70ec5fa0a6f4d617a8b87be802e5c@%3Cusers.subversion.apache.org%3E) (currently) so it fails.

```
svn co https://github.com/libremesh/lime-web/trunk/docs
svn: E170013: Unable to connect to a repository at URL 'https://github.com/libremesh/lime-web/trunk/docs'
svn: E175003: The server at 'https://github.com/libremesh/lime-web/trunk/docs' does not support the HTTP/DAV protocol
```

In order to find a proper alternative I explain the rational behind the original SVN choice in lime-docs Makefile:
in the [lime-web/docs](https://github.com/libremesh/lime-web/tree/master/docs) directory there is a number of documentation files which could be useful for offline, and this is the lime-docs concept.
In order to download all of them without specifying the list of files (requires frequent updating) or to download the whole lime-web, a way was to specify the interesting folder to download in SVN.

Seems that recently [this was implemented also in git](https://github.community/t5/How-to-use-Git-and-GitHub/How-can-I-download-a-specific-folder-from-a-GitHub-repo/m-p/95/highlight/true#M23), and that's great.

I currently lack time to include this in lime-docs, can anyone do this? @aparcar @nicopace @altergui  @G10h4ck @p4u

p4u, 2018-12-18:
Not the most elegant solution but works.

Time to time we will have to "update" the onlydocs branch of git. Maybe would it be possible to create a github hook to do it automatically? Don't know... but for the moment let just fix this blocker issue.

ilario, 2018-12-18:
Github fixed this issue on their side, updating their Subversion implementation, so this issue in fixed (in my opinion).

You can verify this with:

```
svn co https://github.com/libremesh/lime-web/trunk/docs
```

closed 2018-12-18 by ilario

p4u, 2018-12-18:
I don't see this current solution better than the new proposal. But anyway, hope svn is not gonna break again.

ilario, 2018-12-18:
Current solution ([here](https://github.com/libremesh/lime-packages/blob/faed673052653800d3c10be2f62b416bd54f67d4/packages/lime-docs/Makefile#L16-L18)):

. relies on Github subversion support (bad)
. downloads just the needed directory (slightly good)
. always downloads the updated content of lime-web/docs (good)

Proposed solution (#439):

. relies on Github git support (good)
. downloads all lime-web (if I'm correct, slightly bad)
. requires manual intervention each time lime-web/docs contents change (bad)

I'm think we need a third solution, but until then I think the current one is the best option.

p4u, 2018-12-18:
> downloads all lime-web (if I'm correct, slightly bad)
that's not right, it does not download the whole web

> requires manual intervention each time lime-web/docs contents change (bad)
that's right but it also give us the option to specify when we want to "merge" the web documentation to the lime-docs packages.

Anyway, I just want to not see this package breaking libremesh in the future, so any solution is fine.