libremesh/lime-packages #443: Improve and unify GroundRouting documentation

ilario, 2019-01-07:
Users are often confused about what ground routing is and what can be used for (e.g. #367).

We need a good and unified documentation about it.

We can take material from:

* https://github.com/libremesh/lime-packages/blob/master/packages/lime-docs/files/lime-example#L136
* https://github.com/libremesh/lime-packages/blob/master/packages/lime-hwd-ground-routing/Makefile#L27
* https://blog.freifunk.net/author/gmarcos87gmail-com/
* http://old.libremesh.org/projects/libre-mesh/wiki/Ground_routing
* https://webcache.googleusercontent.com/search?q=cache:rrztSEWantwJ:es.wiki.guifi.net/wiki/Supernodos_h%25C3%25ADbridos
* https://docs.google.com/document/d/1tEp4rFNDis4RAwwo4wwg6xDMdqeJKAi8FVYh_FsPEdk
* http://nicolasacco.diveni.re/~gio/internship-report/main.html#libre-mesh

Then generate some good quality documentation on:

https://www.libremesh.org/docs/en_config.html

and reference that page in lime-example, lime-hwd-ground-routing Makefile etc etc.

ilario, 2019-12-29:
Ducking @stale bot, this issue is still valid and require a very little effort to fix!

digitigrafo, 2020-07-17:
ciao, allora c'ho provato, ma non è così facile configurare le vlan. inoltre il router "muore" ogni volta che scazzi e devi riflasharlo.
insomma mi offrirei ma vorrei prima riuscire a farlo andare. AH OPS
HEM I tryed, but is not so easy because vlan. And If you do a mistake you have to reflash the router.. I would like help in this doc but before I need to resolve the config.

Jadhav-Dhanashri, 2026-01-15:
Hello maintainers,
I'm new to freifunk.net open source contributions & excited to get started. I would like to work on this issue, & I'll follow the contribution guidelines carefully. Could you please assign it to me if it's available?
Thank you!