libremesh/lime-packages #458: Sprunge package: rename and use ix.io

ilario, 2019-01-26:
sprunge.us is experiencing some downtime
From sprunge.us maintainer:

> at some point im probably going to deprecate/migrate/close sprunge in favor of ix.io

we should rename the sprunge command to something like `ixio-paste` (by similarity to `lime-paste` in #457) and adapt it to use http://ix.io
The submission method is similar, the only thing that has to be investigated is how to obtain the URL output in the terminal (no idea why it doesn't work as with sprunge.us).

aparcar, 2019-01-26:
Shouldn't we remove this package in favor of lime-paste?

ilario, 2019-01-26:
Yes, basically lime-paste (#457) replaces this package.
So, for the LibreMesh community this package maybe is superfluous, it can be removed from `lime-debug` dependencies.
But it could be useful for the OpenWrt community as an ix.io client or as an alternative to lime-paste in case of problems with paste.libremesh.org domain.
I don't know if something that can be done with sprunge package cannot be done with lime-paste, I think @G10h4ck is currently the only user and should comment here.

ilario, 2019-01-26:
I just managed to understand why the sprunge script was not easy to convert to ix: the ix.io service, for some reason, requires the lines to be terminated in \r\n and not just \n as it was enough for sprunge.
Let me know if this package still fits in lime-packages (in this case I will add the ix client) or if I should move it in my user's repositories.

aparcar, 2019-01-27:
I'd be in favor to remove it, but your choice

ilario, 2019-08-02:
Fixed by #459

closed 2019-08-02 by ilario