libremesh/lime-packages #464: remove libremap legacy

aparcar, 2019-02-03:
The libremesh/libremap repo states that it's outdated. However I can't find a compiled libremap packages anywhere.

Withouth, the lime-full packages is no longer installable

```
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for lime-full:
 * 	libremap-agent
 * 	luci-lib-libremap-location
 * 	luci-lib-libremap-system
 * 	luci-lib-libremap-wireless
 * 	luci-lib-libremap-bmx6
 * 	libremap-agent
 * opkg_install_cmd: Cannot install package lime-full.
make[2]: *** [Makefile:153: package_install] Error 255
make[1]: *** [Makefile:114: _call_image] Error 2
make: *** [Makefile:193: image] Error 2
```

ilario, 2019-02-04:
@nicoechaniz sez: "contesté por algún medio... se puede eliminar."

So yes, we can delete the blocker code.

aparcar, 2019-02-04:
https://github.com/libremesh/lime-packages/pull/465

spiccinini, 2019-02-16:
Already merged

closed 2019-02-16 by spiccinini