libremesh/lime-packages #473: fft-eval: package not arch independent

aparcar, 2019-02-17:
Hi @spiccinini , the fft-eval package isn't arch independent while our CI only compiles for mips. Usually this isn't a problem as we use Lua code, but compiling for every arch a C tool takes to much time. Could you consider creating a PR to add fft-eval to the official OpenWrt repo? Their build bots take care of the rest.

spiccinini, 2019-02-18:
Hi! Another option that I think I prefer is to port the code to Lua, as we only use one function. I really don't know if fft-eval is common enough to add to the official OpenWrt repo, do you know?

aparcar, 2019-02-18:
I'd recommend to push it to the official builds, that's whats supposed to happen with all LibreMesh related packages in some distant future...

gmarcos87, 2019-02-20:
I never did the process of adding a package to openwrt repos. Do you want to try it @aparcar ?

aparcar, 2019-02-20:
https://github.com/openwrt/packages/pull/8265

aparcar, 2019-02-25:
Package was accepted upstream

aparcar, 2019-09-06:
The package is still not available upstream as the compilation requires a special kernel option, which is unlikely to be accepted upstream. Maybe the package shouldn't be a hard dependency for lime-app

gmarcos87, 2019-09-06:
Ok, let me adapt the lime-openairview code so that it only executes the code if the fft-eval package is installed.
I would also like to move fft-eval into a separate repository to keep it available.

aparcar, 2019-09-06:
It's available in the openwrt sources, tho not compiled by default

aparcar, 2019-09-06:
Please let me know once you updated the app. I'll then start building snapshot images.

gmarcos87, 2019-09-09:
@aparcar fft-eval is removed in master now

closed 2019-09-09 by gmarcos87