libremesh/lime-packages #483: DHCP Server on Nodes

kleper, 2019-03-04:
Hello,
I am trying to deactivate the DHCP server of each of the nodes that I have with LibreMESH, I have tried modifying the configuration files of lime-default but something has not gone well and I had to reinstall the firmware.

Thank you.

aparcar, 2019-03-04:
>  But, @kleper, could you re-write the issue on lime-packages which is the proper repo for configuration issues?

Done via this new GitHub feature

ilario, 2019-03-04:
Wowow @aparcar is [the new grua](https://www.tebeosfera.com/publicaciones/no_aparcar_llamo_grua_1991_bou.html)!

ilario, 2019-10-03:
I recall I saw some network-profiles doing this in a brutal brutal way:

https://github.com/libremesh/network-profiles/blob/07eb7730bc8bd66471cfc21d74ab0ab5ef686e5f/trans.hackmeeting.org/L2nodes/etc/uci-defaults/disable_dhcp#L4-L10

https://github.com/libremesh/network-profiles/blob/07eb7730bc8bd66471cfc21d74ab0ab5ef686e5f/trans.hackmeeting.org/L2nodes/PACKAGES#L3-L5

Would it be an ok solution for you @kleper ?

kleper, 2019-10-04:
Yes, this configuration works well for me.
Thank you very much.

closed 2019-10-04 by kleper