libremesh/lime-packages #492: shared-state package requires documentation

natzlob, 2019-03-25:
Hi

I would like to use the shared-state package for sending channel (and other parameter) change information around a bmx6 mesh but I am not clear how to install or use it. Do I need to install all the packages that contain the words shared-state, or just shared-state-bat_host, and then how do I create my own json to send around? 

Please create a README to explain how to use the package.

Thanks

nicopace, 2019-03-25:
it seems there is a commandline tool called shared-state
this one: https://github.com/libremesh/lime-packages/blob/master/packages/shared-state/files/usr/bin/shared-state
ou call it with the data you want to share with others:
https://github.com/libremesh/lime-packages/blob/master/packages/shared-state-bat_hosts/files/usr/bin/shared-state-publish_bat_hosts#L38
`io.popen("shared-state insert bat-hosts", "w"):write(JSON.stringify(batHostTable))`
and you register for this changes by adding a symbolic link to the command that needs to be called with the JSON that was received:
https://github.com/libremesh/lime-packages/tree/master/packages/shared-state-bat_hosts/files/etc/shared-state/hooks/bat-hosts
and you do what you want with that JSON, like in here:
https://github.com/libremesh/lime-packages/blob/master/packages/shared-state-bat_hosts/files/usr/bin/shared-state-generate_bat_hosts

not documentation per se, but hope it can be used as the beginning of one for you, @G10h4ck, @ilario or anyone interested

ilario, 2019-03-25:
@nicopace I never tried shared-state, the only documentation line I found was [this one](https://github.com/libremesh/lime-packages/blob/master/packages/shared-state/Makefile#L27), thanks.

G10h4ck, 2019-03-25:
shared-state distribute data in same broadcast domain. it is not suitable for pure bmx6 mesh, I would go for bmx6-sms in that case

nicopace, 2019-05-29:
@G10h4ck i thought it was using link local, not broadcast, but now I see in the code here https://github.com/libremesh/lime-packages/blob/master/packages/shared-state/files/usr/bin/shared-state-get_candidates_neigh#L42 that you are filtering the neighbours of only the same broadcast domain.
It could be possible that if it detects that it is not in a batman setup, for it to check neighbours over bmx and not linklocal, so it could be compatible with this setup.
Of course, PR welcomed :)

ilario, 2019-08-17:
5 months have passed, has some documentation been written?
I think I managed to use it in a basic way, but it was not easy: documentation is really needed!!!

nicopace, 2019-08-25:
@gmarcos87 did some documentation in spanish of it:
https://mesheando.net/tutoriales/shared-state/

javierbrk, 2024-11-21:
I have been writing some documentation in specific packages and also in shared-state-async lime packages readme. 
I can improve it and propose more detailed information. So far the best we have is: 

https://github.com/libremesh/lime-packages/blob/08784318f4e9fd4269675bd9dbc8ebf6962ce5da/packages/shared-state-ref_state_commons/README.md

https://github.com/libremesh/lime-packages/blob/master/packages/shared-state-async/README.adoc

ilario, 2024-11-23:
Fixed by #1142 which added documentation to this file: https://github.com/libremesh/lime-packages/blob/master/packages/shared-state-async/README.adoc

Thanks @javierbrk !!