libremesh/lime-packages #501: Chef not compiling correct firmware

tomjones1977, 2019-04-19:
Creating what should be a fairly straightforward build with Chef seems to fail.

Settings:
Distribution - Libremesh
Version - master
Device model - Ubiquiti Nano M XW (ar71xx/generic)
Flavour - LibreMesh with BMX7
Network Profile - libremesh-default

Upgrading the NanoStation loses the anygw functionality, the LiMe web interface, and I can no longer include wpad-mesh as it isn't found, meaning my mesh settings no longer work either.

Help!

nicopace, 2019-05-29:
@aparcar, I think the experience of the users started breaking when the default config in chef is bmx7only images, as most of the libremesh config asumes batman is present.
could you put batman+bmx6 as a default till someone puts energy in some other experience?
we are working on babel integration, but it is still a WIP.
the other option is to stop promoting chef and just give a default image (now that firstbootwizard exist, basic customization is no longer required).
one way or the other, newcomers have a terrible experience, so chef is worse than not having chef in the current situation.

aparcar, 2019-08-12:
Please create a PR for Chef, whatever flavor you want is fine with me.

aparcar, 2019-08-12:
>  one way or the other, newcomers have a terrible experience, so chef is worse than not having chef in the current situation.

I once deactivated LibreMesh from Chef as the experience is that terrible, I was told to put it back there by an LiMe dev. Anyway, as I'm currently working on lower OpenWrt layers, I'm fine with anything you find useful.

ilario, 2019-10-03:
Chef is temporarily (?) not including compilation options for LibreMesh. Closing.

closed 2019-10-03 by ilario