libremesh/lime-packages #512: File collisions and dependency hell

dangowrt, 2019-05-15:
cooker failes complaining about both `dnsmasq-lease-share` and `shared-state-dnsmasq_leases` trying to install the same files:
```
 * check_data_file_clashes: Package dnsmasq-lease-share wants to install file /usr/src/lime-sdk/snapshot/ramips/mt7621/ib/build_dir/target-mipsel_24kc_musl/root-ramips/etc/uci-defaults/90_dnsmasq-lease-share
	But that file is already provided by package  * shared-state-dnsmasq_leases
 * check_data_file_clashes: Package dnsmasq-lease-share wants to install file /usr/src/lime-sdk/snapshot/ramips/mt7621/ib/build_dir/target-mipsel_24kc_musl/root-ramips/usr/bin/dnsmasq-lease-share.sh
	But that file is already provided by package  * shared-state-dnsmasq_leases
 * opkg_install_cmd: Cannot install package lime-full.
```

It wasn't that trivial for me to solve due to hard-dependencies introduced by the `lime-full` meta-package (which imho should be flattened-away and just use flavors instead, providing soft-dependencies instead).

ilario, 2019-05-15:
> [lime-full meta-package] should be flattened-away and just use flavors instead

Totally agree on that!!!

Just I have the doubt on what happens if we decide to support again the buildroot method (with `make menuconfig`) rather than lime-sdk: could we use flavors in that or we would need the meta-packages?

dangowrt, 2019-05-15:
@ilario we could easily provide a script to modify the buildroot's `.config` and select the packages of a flavor. The selection can then still be altered by the user using `menuconfig`.

ilario, 2019-05-16:
Like that!
In case people using buildroot wanted to keep the meta-packages, we could still remove them from the flavors (replacing lime-full with the complete list of the packages it should pull).

G10h4ck, 2019-05-16:
I use the build-root but never liked those glue-packages, I am all to have them removed, the laziness to not read a list of packages in the documentation and select them depending on one needs, is not a good reason to keep that maintaining hell cruft around IMHO

G10h4ck, 2019-05-16:
also `dnsmasq-lease-share` and `shared-state-dnsmasq_leases` should never be installed together, IMHO the latter is the one that works out of the two

ilario, 2019-05-26:
So, `lime-full` is gone. 
Do we want to eliminate also `lime-basic`?
For each package we eliminate, we will have to remember to fix what was referencing to it.

G10h4ck, 2019-05-26:
> Do we want to eliminate also `lime-basic`?

+1

ilario, 2019-10-03:
We should update lime-basic (see #607) or delete it. Anyway the "dependency hell" is solved. Closing.

closed 2020-01-04 by ilario