libremesh/lime-packages #519: Piranha is blocking thisnode.info requests

nicopace, 2019-06-04:
**What I expect to see**: in a newly flashed device with piranha package installed, if I go to thisnode.info I expect to see the luci interface

**What I actually see**: It redirects to piranha page.

luandro, 2019-06-11:
If you're using https://github.com/LibreRouterOrg/openwrt it doesn't come with luci, that might be why.  The images we use in Moinho we've added luci and thisnode.info works fine.

G10h4ck, 2019-06-11:
@nicopace as asked earlier in chat, did you tested if it is a matter of thisnode.info being resolved to wrong IP?

nicopace, 2019-06-12:
nope, we couldn't test.
will try to test again, as well as the luci thing, but can't promise soon.

gmarcos87, 2019-09-05:
@luandro any new? Or we can close this

luandro, 2019-09-05:
Guess we can close right? What happens and is expected is that thisnode.info is directed to Lime App.

closed 2019-09-05 by luandro