libremesh/lime-packages #521: shared-state is not proposing a neighbour in one direction, but it is in the other

nicopace, 2019-06-08:
In QuintanaLibre, happens the following
```bash
nico@ingorian ~> ssh -A root@thisnode.info
root@ql-roxa:~# shared-state-get_candidates_neigh
fe80::96:66ff:fef1:5251%wlan1-adhoc_29
fe80::eade:27ff:fe91:606e%wlan1-adhoc_29
root@ql-roxa:~# ssh -A fe80::96:66ff:fef1:5251%wlan1-adhoc_29
root@ql-nicojesigioia:~# shared-state-get_candidates_neigh 
fe80::e1:d3ff:fe19:4e52%wlan2-adhoc_29
fe80::eade:27ff:fe91:606e%wlan1-adhoc_29
root@ql-nicojesigioia:~# ssh -A fe80::e1:d3ff:fe19:4e52%wlan2-adhoc_29
root@ql-oncelotes-bbone:~# exit
root@ql-nicojesigioia:~# ssh -A fe80::eade:27ff:fe91:606e%wlan1-adhoc_29
root@ql-gioiajesinico:~# exit
root@ql-nicojesigioia:~# exit
root@ql-roxa:~# shared-state-get_candidates_neigh 
fe80::96:66ff:fef1:5251%wlan1-adhoc_29
fe80::eade:27ff:fe91:606e%wlan1-adhoc_29
root@ql-roxa:~# ssh fe80::96:66ff:fef1:5251%wlan1-adhoc_29
root@ql-nicojesigioia:~# shared-state-get_candidates_neigh 
fe80::e1:d3ff:fe19:4e52%wlan2-adhoc_29
fe80::eade:27ff:fe91:606e%wlan1-adhoc_29
```

shared-state-get_candidates_neigh from roxa shows ql-nicojesigioia, but ql-nicojesigioia doesn't show roxa.

gmarcos87, 2019-06-08:
Whats are the fe80 addresses of ql-roxa?

nicopace, 2019-06-08:
don't know... but the mac addresses of it are:
```
root@ql-roxa:~# iw dev | grep addr
		addr 14:cc:20:ad:b0:6b
		addr 16:cc:20:ad:b0:6a
		addr 14:cc:20:ad:b0:6a
```

gmarcos87, 2019-06-08:
fe80::14cc:20ff:fead:b06a is filtred becose include ::1. Its a bug in shared-state-get_candidates_neigh

nicopace, 2019-06-08:
```
root@ql-roxa:~# ip -6 a | grep wlan -A 1 | sed 's/@.*$//g'
16: wlan0-adhoc: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1536 state UP qlen 1000
    inet6 fe80::16cc:20ff:fead:b06a/64 scope link 
--
18: wlan0-apname: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 state UP qlen 1000
    inet6 fe80::14cc:20ff:fead:b06a/64 scope link 
--
21: wlan0-adhoc_17
    inet6 fe80::16cc:20ff:fead:b06a/64 scope link 
--
22: wlan0-adhoc_29
    inet6 fe80::16cc:20ff:fead:b06a/64 scope link 
--
30: wlan1-adhoc: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1536 state UP qlen 1000
    inet6 fe80::16cc:20ff:fead:b06b/64 scope link 
--
31: wlan1-adhoc_17
    inet6 fe80::16cc:20ff:fead:b06b/64 scope link 
--
32: wlan1-adhoc_29
    inet6 fe80::16cc:20ff:fead:b06b/64 scope link 
```

spiccinini, 2019-06-15:
Closing as it is fixed!

closed 2019-06-15 by spiccinini