libremesh/lime-packages #527: lime-apply doesn't apply changes on batman-adv vlan change

nicopace, 2019-06-11:
Expected behaviour: after changing batman-adv vlan, and doing `lime-config; lime-apply`, for the vlan to change
Actual behaviour: the vlan doesn't change till you reboot.

Proposed implementation options:
* to reboot the node
* to notice the user that that change won't be applied till reboot
* to reload the module
* any other option?

ilario, 2019-10-03:
`lime-apply` is currently useless as it is extremely outdated.
We should either include more stuff or delete it.
Regarding the VLAN change, I suppose (but I didn't test) that a 
`/etc/init.d/network restart`
could do it.

gmarcos87, 2019-10-03:
I can't reproduce this bug, lime-config and lime-apply change batman vlan successfully

ilario, 2019-10-04:
Ok, closing.
On lime-apply being useful or not, see discussion on #352.

closed 2019-10-04 by ilario