libremesh/lime-packages #540: Local .lan domain not easy to use on mobile and on Chrome/Chromium

ilario, 2019-06-22:
The currently in use local domain (FQDN) `.lan` works perfectly when using Firefox on desktop or on mobile.
As [reported by a user on lime-users mailing list](https://lists.libremesh.org/pipermail/lime-users/2019-May/001457.html), on most of the other browsers (I tested Chrome, Chromium, Opera, Samsung Internet Browser) either on desktop or on mobile, when typing `lime-ddeeff.lan` this string is searched on search engines rather than opened as an URL.

This is easy to workaround either adding a slash `lime-ddeeff.lan/` or specifying the protocol `http://lime-ddeeff.lan`, still this can be a problem for most of the users.

Two possible solutions have been proposed in the mailing list:

* use an accepted but abandoned top level domain (to be tested .bv .sj .iq .um .su .gb .gu .kp)
* use an existing and reliable domain (like `thisnode.info` which is owned by Altermundi community network and is already used for connecting to the closest network node) which would result in something like `lime-ddeeff.thisnode.info`

nicopace, 2019-06-22:
Altermundi also owns the libre.social domain, that i think it is much better suited for this.
There was an approach for this already implemented... can you remember @nicoechaniz @altergui ?

ilario, 2019-06-23:
@nicopace In #541 I suggested `thisnode.info` also because users already know and remember that, as it is already in use.

gmarcos87, 2019-09-05:
When I was watching a libremesh workshop in Brazil I noticed that they were teaching how to access with the url **nodename.lan**.
In my case I never used because I always specified my own domain in /etc/lime/config or I access through **http://nodename**
I mean that there are already people accustomed to .lan, which being "neutral" in different languages and is definitely shorter.
The only point in favor of thisnode.info is that we can show a message if it is accessing through the 3G connection of a cell phone instead of WiFi. As thisnode.info is the domain of anygw we can add that feature anyway.

Researching a little I came to this Chromium Wontfix https://bugs.chromium.org/p/chromium/issues/detail?id=30636
Instead of adding http:// at the beginning it is only necessary to add one / at the end. A sad decision that only benefits Google.

commit 5378b10e (2019-11-12): Merge pull request #541 from ilario/fqdn

Replaces .lan local domain with .thisnode.info fixes #540
closed 2019-11-12 by G10h4ck