libremesh/lime-packages #542: lime wireless channel config inconsistent order

spiccinini, 2019-06-22:
We found that at least in one LibreRouter device the channel config is not applied in the same order as the other. So for example wlan1-mesh is channel 157 and not 48 like all the other devices I have seen. Both /etc/config/lime are identical. PCI devices are enumerated in the same way.

This is the part of the config:
```
	list channel_5ghz '157'
	list channel_5ghz '48'
```

gmarcos87, 2019-06-22:
We need to look at how channel options are being assigned. I also find it not very intuitive in the case of multiple options and a single radio.

G10h4ck, 2019-06-23:
Which channel get each radio is calculated here https://github.com/libremesh/lime-packages/blob/master/packages/lime-system/files/usr/lib/lua/lime/wireless.lua#L117

Basically it depends on the number of the iface so wlanX will take channel that depends on X and how much channels we have in the list of channels `local chanIndex = 1 + radioName:match("%d+$") % #channel` so the order in the list is important too.

Running again `lime-config` solved the issue or it is happening always with the same router?

ilario, 2019-08-01:
The channel which get picked from the list depends on the "radio0", "radio1" assigned names.
Do we know how radioX names gets assigned?
For sure, if a different number of radios is present in one of the routers (e.g. missing the 2.4 GHz one), the names could shift.
If the radioX name is not reliable, what could we use?

ilario, 2019-10-03:
@spiccinini can you check if, using that same router, the problem is still present even after the merge of #603?

spiccinini, 2019-10-07:
I don't haver that router at hand anymore and also reproducing this was not consistent. The other it happened and then I tried to reproduce it in many ways (like doing a first_boot) but I could not.

ilario, 2019-10-08:
Mmmh... Do you think we can close this?

spiccinini, 2019-10-08:
Yes, I will reopen if I see it again with the ñatest code.

closed 2019-10-08 by spiccinini

spiccinini, 2019-11-20:
Hmmmm.... now I realize that maybe this was a side effect of FirstBootWizard changing the frequency  one of the radios to the other librerouter radio frequency!

spiccinini, 2019-11-29:
> Hmmmm.... now I realize that maybe this was a side effect of FirstBootWizard changing the frequency one of the radios to the other librerouter radio frequency!

Ok, now I confirm that this was the issue... so it wasn't an issue at all.... it's just that I didn't knew at the moment that firstbootwizard would be changing channels (and changing the config/wireless to do so).