libremesh/lime-packages #558: How to use network-profiles with OpenWrt buildroot compilation method?

ilario, 2019-08-12:
The suggested compilation method has changed over the time (buildroot, lime-build, lime-sdk, buildroot).

With the LibreMesh 17.06 release, communities accepted to use [flavors](https://github.com/libremesh/lime-sdk/blob/develop/flavors.conf) (e.g. lime_default, lime_mini, lime_zero...) and [network-profiles](https://github.com/libremesh/network-profiles) which nicely work with [lime-sdk](https://github.com/libremesh/lime-sdk) and [Chef](https://chef.libremesh.org/).

While Chef is still supported, [searching into the mailing list](https://lists.libremesh.org/pipermail/lime-users/2019-May/001467.html) seems that the currently supported local compilation method, changed from lime-sdk to the OpenWrt buildroot. 

The documentation about LibreMesh compilation with OpenWrt buildroot ([here](https://github.com/LibreRouterOrg/openwrt/blob/librerouter-18.06/README.md) and [here](https://github.com/libremesh/lime-packages/blob/master/README.md)) does not mention neither flavors nor network-profiles.

The buildroot method documentation has to be expanded to help the lime-sdk => buildroot transition for the communities.

ilario, 2019-08-14:
The mechanism employed by https://github.com/aparcar/openwrt-metabuilder is to install the network-profiles as packages downloading them from here http://chef.libremesh.org/network-profiles/ or here http://repo.libremesh.org/network-profiles/ I think we should chose one of these and suggest to the users to use it.
The Chef's one?
@aparcar @p4u

spiccinini, 2019-08-15:
> While Chef is still supported, [searching into the mailing list](https://lists.libremesh.org/pipermail/lime-users/2019-May/001467.html) seems that the currently supported local compilation method, [...]

Thanks Ilario for gathering all this information. I didn't know about network_profiles_builder. 

To support network-profiles in _the buildroot way_ I am inclined to do it with a symbolic link to a local network-profiles repository:
```
$ cd openwrt
$ ln -s ~/path/to/network-profiles/networkname/device/ files
```
So it is easy for the developer to change the network-profile and rebuild and it is also the way that the buildroot expects to add files to the build. The only change required is to add this step to the documentation.

What do you think?

ilario, 2019-08-15:
About network-profiles, is it possible to add a binary repository to the buildroot?
Something like:

```
src/gz lm_profiles http://chef.libremesh.org/network-profiles/
```

---

In my opinion flavors can be abandoned, people using buildroot can select the list of the packages they need and who uses the [metabuilder](https://github.com/libremesh/lime-packages/pull/560) can customize the example files with a default list of packages.

spiccinini, 2019-08-15:
> src/gz lm_profiles http://chef.libremesh.org/network-profiles/

As far as I know it is not posible to use 'binary feeds'.

ilario, 2019-08-15:
Ah...
One thing that the packages do and the `files` directory does not do is to manage the dependencies included in the network-profiles (e.g. a community can select just batman-adv packages and nothing else, and doing so selecting the packages in the profile like [here](https://github.com/libremesh/network-profiles/blob/master/trans.hackmeeting.org/L2nodes/PACKAGES)).
@aparcar does [network_profiles_builder](https://github.com/libremesh/network-profiles-builder) create a sources repository?

ilario, 2019-10-03:
So, network_profiles_builder do create valid makefiles and packages' sources.
In order to use them as a feed, we need to put these packages in a git repository (the feeds have to be either on the local drive or, if they are online, on a control version system) and to copy this on some server. This can be done via a continuous integration bot.
The only bit of the network-profiles that will not work this way, is the negative selection of packages (removing unwanted packages).

ilario, 2020-01-09:
The implementation in the continuous integration is still missing.

ilario, 2020-01-10:
This is a duplicate of https://github.com/libremesh/network-profiles-builder/issues/2 on the network-profiles-builder repository. Closing here, using the other issue for keeping track of the work to do.

closed 2020-01-10 by ilario