libremesh/lime-packages #575: lime-ap-watchping fail after reboot

gmarcos87, 2019-09-06:
Reading the lime-ap-watchping code I notice that the original ap ssid is stored in /tmp waiting for the internet ping to return (https://github.com/libremesh/lime-packages/tree/master/packages/lime-ap-watchping/files/etc/watchping/autoAP-fail.d)
In the process the /etc/config/wireless configs are replaced.

The problem is that if the node is restarted when it is "down" it loses the original apssid. At that 
reboot you may detect again that there is no internet and add "-down" to the apssid again and again.

I haven't tried it, but that seems to be the way it works. Does anyone use this package? Can you check this bug?

ilario, 2019-10-03:
Funnily enough, the script sets an UCI option but the `uci commit` command is missing.
So the new option is being applied by the `wifi` command but it does not get saved in the `/etc/config/wireless`.
Can you confirm? Can we close this?
Anyway I suspect nobody is using this package.

ilario, 2020-01-09:
As explained in my comment above, the AP name from the configuration file does not seem to be removed. So this would not be an issue at all.
Due to the lack of either confirmation of the information backing this issue, I'm closing.

closed 2020-01-09 by ilario