libremesh/lime-packages #582: shared-state waits for uhttpd for too long

spiccinini, 2019-09-09:
Sometimes uhttpd accepts the connection but never responds so shared-states doesn't timeout quickly. Restarting uhttpd fix this at least temporarly.

We may think about changing uhttpd + cgi + bash + lua for something else? Maybe OWSD tiny (that has ubus support) ?

```
openwrt version 18.06.4
LIME_REVISION="ec81de9"
LIME_DESCRIPTION="LiMe master development (master rev. ec81de9 20190613_1242)"
LIME_BUILDDATE="20190613_1242"
```

ilario, 2019-10-04:
There is some timeout option in shared-state [here](https://github.com/libremesh/lime-packages/blob/master/packages/shared-state/files/usr/lib/lua/shared-state.lua#L182-L183), maybe we should check out if we miss some more timeout option?
Otherwise, we can kill the process after some time, I did this in a very dirty way in bandwidth-test package [here](https://github.com/libremesh/lime-packages/blob/31c5d92f3c316983479afbf3da1042beec809305/packages/bandwidth-test/files/bin/bandwidth-test.lua#L78-L82).

spiccinini, 2019-10-07:
> There is some timeout option in shared-state [here](https://github.com/libremesh/lime-packages/blob/master/packages/shared-state/files/usr/lib/lua/shared-state.lua#L182-L183), maybe we should check out if we miss some more timeout option?

The problem seems to be that the response is very slow after the socket connection is made so this options doesn't help in this case.

> Otherwise, we can kill the process after some time, I did this in a very dirty way in bandwidth-test package [here](https://github.com/libremesh/lime-packages/blob/31c5d92f3c316983479afbf3da1042beec809305/packages/bandwidth-test/files/bin/bandwidth-test.lua#L78-L82).

Yes I think that we must kill the connection somehow if it is taking too long. I don't know what is the best way.

ilario, 2019-10-11:
Looks like this is being addressed in #622 and #623

spiccinini, 2019-11-21:
The correct fix was done here https://github.com/LibreRouterOrg/openwrt/commit/c9dad82ea2c922d8be1e17a159fd477bfeb2e6af

Sadly upstream did not take the patch yet (I sent it a month ago to openwrt list and to uhttpd developers), here is the bug I opened https://bugs.openwrt.org/index.php?do=details&task_id=2562

ilario, 2020-01-09:
Is there any way to push this forward?

spiccinini, 2020-09-16:
It is fixed in https://git.openwrt.org/?p=project/uhttpd.git;a=commit;h=939c281c70ef82929ffcb3cb383fd6fdc40f341b

closed 2020-09-16 by spiccinini