libremesh/lime-packages #589: FirstBootWizard: doesn't see a network

nicopace, 2019-09-10:
What I expect to happen:
* when I open the FirstBootWizard I want to see the existing networks around me. In particular, from ql-roxa want to see ql-nicojesigioia

What I see:
* If I ssh to the node, I can see the adhoc networks provided by the node. If I directly connect to the neighbouring node, I can see that is sharing its lime-defaults file. but if I enter the lime-app firstbootwizard screen it doesn't show this node as an option (it actually says there is noone around).

How to reproduce it:
* flash a node with the current librerouterorg/openwrt image of libremesh: point it to ql-nicojesigioia

gmarcos87, 2019-09-11:
Nico, I told you, you can't try fbw with quintanalibre. For many reasons, but surely what happened to you is related to the use of TZ channels on the network and not on the new device.

nicopace, 2019-09-11:
Awesome, just wanted to keep track of the issue so we were not forgetting it (mainly I)... will dig into it tomorrow...

markbirss, 2019-11-13:
@gmarcos87 this relates to my email sent, I will ask here about my issue

config-sharer-server package ? this is misspelled ?

I have compiled a fresh openwrt image from source added lime-app on 2 nodes
created a network on the 1 device but the other device cannot find the created network

spiccinini, 2019-11-14:
@markbirss Which device are you using? AFAIK firstbootwizard only searches 5GHz mesh networks.

markbirss, 2019-11-14:
@spiccinini
Ok, thank you for your feedback,  will check again with couple Aruba AP-105 and Meraki MR24's

was using a Aruba AP-105 (2.4/5Ghz) and GLI-Net USB150 (2.4Ghz)

regarding that package what is config-sharer-server ?

markbirss, 2019-11-15:
Ok, I'm still having issue after creating a configured node not being able to scan for it. It seems the configuration is somehow broken.

ilario, 2020-09-22:
I suppose this can be explained with #661 so I will close this as a duplicate.
Feel free to re-open it if you disagree! :)

closed 2020-09-22 by ilario