libremesh/lime-packages #596: Babeld without VLAN does not run on LAN ethernet ports

ilario, 2019-09-18:
Blocks #581 

tl;dr without VLAN, Babeld currently cannot run on ethernet

When instructing Babeld to run on the plain interfaces, without any VLAN, it fails to run on LAN ethernet ports.
The reason is that they lack an IPv6 Link Local address, and they cannot have it as they are included in `br-lan` bridge.

So, just in the case of Babeld witout VLAN, we should run it directly over the br-lan bridge.

The problem is that Batman-adv interface is also into the bridge and to have hello packets of Babeld running into Batman-adv would completely break Babeld topology estimation.

So some firewall rules like the ones in lime-proto-bmx6 [here](https://github.com/libremesh/lime-packages/blob/master/packages/lime-proto-bmx6/src/bmx6.lua#L130-L134) will have to be used.

ilario, 2019-09-18:
Seems that we could filter packages from and to port 6696.
https://wiki.reseaulibre.ca/documentation/babel/#firewallconfiguration

Do you think that this solution would be too complex to be active by default?
I mean, there is some interest (see #581) to remove the default VLAN from Babeld, do you think it is still worth, now that we discovered this complication?

spiccinini, 2019-09-19:
Thanks @ilario for pointing this out! If it was working good in 17.06 its ok for me using the firewall, I don't know what complications could arise. @G10h4ck  ?

ilario, 2019-09-25:
I started to get it working in #600. It is not ready to merge but it served well for a 3 days event with 70 participants :)