libremesh/lime-packages #597: Consider devices with WAN ethernet port only or no ethernet port

ilario, 2019-09-18:
See comments on #595 

I thought this solution, but maybe we'll never use such a device or this is not a good solution?

In network.lua
```
-               ifname = board['network']['lan']['ifname']
+               local lanTable = board['network']['lan']
+               local wanTable = board['network']['wan']
+               if lanTable then
+                       ifname = lanTable['ifname']
+               elseif wanTable then
+                       ifname = wanTable['ifname']
+               end
```

wfleurant, 2019-09-20:
Unifi AC Mesh is popular (not just a shameless plug heh)
- https://docs.karlsruhe.freifunk.net/firmware/unifi-ac/
- https://massmesh.org/wiki/index.php?title=Ubiquiti_UAP_AC_Mesh

Some times I want eth0 as a LAN and sometimes as a WAN. This is just a fix for nil table, no?

ilario, 2019-09-21:
GL-iNet AR300M-Lite has just a WAN port https://openwrt.org/toh/gl.inet/gl.inet_gl-ar300m-lite#specific_configuration

ilario, 2024-11-11:
> Some times I want eth0 as a LAN and sometimes as a WAN. This is just a fix for nil table, no?

Sorry @wfleurant fir answering after 5 years XD

No, this is for dealing with hardware with no port labelled as LAN. From the OpenWrt wiki, seems that Unifi AC Mesh has a port recognized by OpenWrt as LAN.

For configuring a port as WAN, follow this example: 
https://github.com/libremesh/lime-packages/blob/6642f373908d67c9cfc9d8f360843e9c528a8b94/packages/lime-docs/files/www/docs/lime-example.txt#L217-L224

Regarding the issue: the problem should still be possible, but nobody reported it in years. Not worth to fix. Closing until when someone reports it.