libremesh/lime-packages #602: List of 5 GHz channels and one or two radio devices

ilario, 2019-09-26:
The current [implementation of the list of channels](https://github.com/libremesh/lime-packages/blob/69ed489fc771cbd3bd4e0047ba2c926e98304ce8/packages/lime-system/files/usr/lib/lua/lime/wireless.lua#L119-L122) (you can specify a list of wifi channels instead of a single channel, then each radio on the same band should receive a different channel, [currently this is active by default](https://github.com/libremesh/lime-packages/blob/69ed489fc771cbd3bd4e0047ba2c926e98304ce8/packages/lime-system/files/etc/config/lime-defaults-factory#L47-L48)) is troublesome when considering networks with mixed one and two radios devices.

Let's take an example using the current default configuration and two commonly employed routers:

* LibreMesh on TP-Link WDR3600: `wlan0` has channel 11, `wlan1` has channel 48
* LibreMesh on Ubiquiti NanoStation M5: `wlan0` has channel 157 (if country code is ok, see #594)

one would expect the two routers to connect each other, but this does not happen because of the channels.
Is it intentional? To me it looks like a bug.

G10h4ck, 2019-09-26:
can you show the configuration file?

ilario, 2019-09-26:
> can you show the configuration file?

I don't have the routers with me now, but the lime-defaults was the one from [lime-defaults-factory](https://github.com/libremesh/lime-packages/blob/7cd89460c80008e2cc7d2f1f539cbb0c3f87ece8/packages/lime-system/files/etc/config/lime-defaults-factory#L47-L48) and the example used above is exactly what happened.
This is not surprising as the implementation is using the `wlan#` number for indexing the channels list.

G10h4ck, 2019-09-26:
With the current implementation either you use radio specific config section, or it behave like you described

ilario, 2019-09-26:
Yess, what I mean is that with the _default_ configuration the behaviour should not be this weird.

So, IMHO either we change the implementation of the channels list or we leave just one channel in the _default_ configuration.

ilario, 2019-09-28:
@spiccinini @G10h4ck do you think we can implement this feature in a better way?
For example considering either the 5 GHz or the 2.4 GHz radios instead of all of them?

spiccinini, 2019-09-28:
Yes, I think a better implementation is to count independently the radios in each band.

PR #603 (2019-10-02): Simpler WLAN channel config
Fixes #602 . Also related to #503 
This is not backward compatible as the channel list must be reordered prior to upgrade. I am not very happy with this but the current config is odd so....  One way to do it backward compatible is to add a new option like simple_channel_sorting = 1 and use this code only when the option is present (or something like that) but I don't know as it will add more complexity.

What do you think?
closed 2019-10-02 by gmarcos87