libremesh/lime-packages #604: Do we need dnsmasq "local=/domainName/" option?

ilario, 2019-09-30:
What happened to me was:
building the network for an event, I used their website name, e.g. `mywebsite.org`, as a domain for the network and the mywebsite.org website was not accessible anymore.

I am not sure this is the correct behaviour (and I should have been using something else as a domain, like `mesh.mywebsite.org`) or if it is an issue to be fixed.

If one sets the option `domain=mywebsite.org` in `/etc/config/lime`, then the `local=mywebsite.org` [local option](http://thekelleys.org.uk/gitweb/?p=dnsmasq.git;a=blob;f=man/dnsmasq.8;h=bc5ae6360f5459d99cafb78d34c532e5b087abf6;hb=HEAD#l426) appears in the dnsmasq configuration. This happens in [network.lua](https://github.com/libremesh/lime-packages/blob/master/packages/lime-system/files/usr/lib/lua/lime/network.lua#L130).

With this option `mywebsite.org` and `*.mywebsite.org` are just resolved locally (the external DNS servers are not used for these domains) which means that if mywebsite.org is not hosted inside the mesh, it cannot be accessed.

So currently one should not specify `mywebsite.org` but rather `mesh.mywebsite.org` there.

This is correctly documented [in lime-example](https://github.com/libremesh/lime-packages/blob/master/packages/lime-docs/files/lime-example#L19) but incorrectly documented [on the website](https://github.com/libremesh/lime-web/blob/617896409cc1a2a2806554356a1d4dcc74abc4e4/docs/en_config.txt#L53).

Is the current behaviour the correct one and should we update the documentation on the website _or_ there's no problem to use `mywebsite.org` as a domain and we should remove that `local` option?

G10h4ck, 2019-10-01:
The documentation on the website is wrong, the one in lime example ok, something like mesh.yourcommunity.net is for what it is intended for that option.

This kind of things happens a lot when the documentation is duplicated around, that is one of the skepticisms I have about creating more documentation platrorms, like github wikis, docuwiki, mediawiki, README etc, the ultimate documentation is in the code, so if we want to expose that on the web we should use something like doxygen that extract the documentation directly from the code using inline comments + pages.

ilario, 2019-10-01:
> The documentation on the website is wrong, the one in lime example ok, something like mesh.yourcommunity.net is for what it is intended for that option.

Ok, see 

> This kind of things happens a lot when the documentation is duplicated around, that is one of the skepticisms I have about creating more documentation platrorms, like github wikis, docuwiki, mediawiki, README etc,

I agree.

> the ultimate documentation is in the code, so if we want to expose that on the web we should use something like doxygen that extract the documentation directly from the code using inline comments + pages.

I disagree.
An example of what you say is #573, we'll see how it goes.

gmarcos87, 2019-10-02:
In #573 I'm not trying to generate documentation for the end user but for developers, or even for ourselves. In my opinion, it is easier to approach the code of a package if there is something in the readme. It also helped me to start a path of cleaning and revision that ended with several deprecated packages.
It is not the intention of that documentation to replace what exists in lime-web, it is another purpose.
I also find it more comfortable that when making changes to the code of a package also keep the documentation updated in the same pull-request.

ilario, 2019-10-03:
It's amazing that there is material for discussion, as we have three different opinions.
But let's do it on lime-users mailing list which is the right place and let's close this issue which has been solved.

closed 2019-10-03 by ilario