libremesh/lime-packages #608: Update build instructions on readme

ilario, 2019-10-03:
In the readme the suggested compilation methods are outdated.

Chef is temporarily not supporting LibreMesh.
lime-sdk is broken by various issues and core developers are not maintaining it anymore.
OpenWrt buildroot instructions should be removed and just a link to the [development page on the website](https://libremesh.org/development.html) should be present there.

We should consider whether to add the @aparcar's [metabuilder](https://libremesh.org/development.html) between the official compilation methods.

bharrod321, 2019-10-05:
I made the requested change above and have a pending pull request.  This is my first time doing something like this.  Please let me know if I did anything incorrect.

Thank you.

ilario, 2019-10-06:
Thanks @bharrod321!!
May I ask you the following changes?

* lime-sdk is not recommended anymore, can you remove the phrase "either using the easy to use
[lime-sdk][2] tool (recommended) or manually " and also the link number 2 at the bottom of the document?
* can you remove also the whole section titled "### Using lime-sdk"?
* in the development page on the website there are updated instructions for the build, so could you remove also the whole "### Using OpenWrt buildroot" section?
* the link to the development page on the website, you included it in the right place, but could you include it using the syntax used for the rest of the Readme (which is: `[some text bla bla bla][123]` and the link at the end of the page `[123]: https://libremesh.org/development.html`)

Thanks again!!

bharrod321, 2019-10-12:
I have created a PR with the above mentioned changes.  Sorry it took so long to get back to you.

ilario, 2019-10-13:
#627 fixed

closed 2019-10-13 by ilario