libremesh/lime-packages #609: Redirect from LuCI to lime-app not working

ilario, 2019-10-04:
I am not experienced with lime-app, but when I open thisnode.info what I get is just this:

[image]

and I have to explicitly include `/app` in the URL in order to get the FirstBootWizard or the app.
What am I doing wrong?
I compiled following the instructions [on the website](https://libremesh.org/development.html).
Thanks!

ilario, 2019-10-04:
I have luci-basic as it is a dependency of `luci-lib-libremap` which is selected by `libremap-agent` which is selected by `lime-map-agent` (see also #433), could it be the problem?

spiccinini, 2019-10-07:
> If I am correct, lime-app pulls a cascade of dependencies which includes luci-basic...?
Do you mean `luci-base`?  

How did you build the image? We are adding  this file to the buildroot `files/www/index.html`
``` 
<?xml version="1.0" encoding="utf-8"?>
<!DOCTYPE html PUBLIC "-//W3C//DTD XHTML 1.1//EN" "http://www.w3.org/TR/xhtml11/DTD/xhtml11.dtd">
<html xmlns="http://www.w3.org/1999/xhtml">
<head>
<meta http-equiv="Cache-Control" content="no-cache" />
<meta http-equiv="refresh" content="0; URL=/app" />
</head>
<body style="background-color: white">
<a style="color: black; font-family: arial, helvetica, sans-serif;" href="/app">LiMe-App</a>
</body>
</html>
```

ilario, 2019-10-08:
> How did you build the image?

https://libremesh.org/development.html

>We are adding this file to the buildroot files/www/index.html

Ah!!!!!!! I didn't know!
I realised that [you're adding also lime-defaults-factory](https://github.com/LibreRouterOrg/openwrt/blob/0cb606f0293b257ec6cbf3b1013ae777ccfea82e/files/etc/config/lime-defaults-factory), can't you just use the one in lime-system?

> Maybe we may want to add this file in the lime-app package?

Yes!!!!!!!!!!!

> I think that if you also select a full luci installation this will be a problem as both packages will provide the same file (but I am not very sure)

Yes, actually lime-basic has the same file, so this is a problem as it would generate a conflict between the lime-basic and the lime-app packages.

ilario, 2019-10-11:
I see three possible solutions:

* the dirty one is to include an alternative index.html (as in the LibreRouter repository) but inside some script that installs it on the first boot (just including it in lime-app would conflict with luci-basic)
* we fix the dependency chain (I described it in a comment above) that makes lime-app depend upon luci-basic (and this would solve also #433)
* we set the `uhttpd.main.home='/www'` option to be `uhttpd.main.home='/www/app'` and we replace the mentions to `/app` with just `/` in lime-app code

ilario, 2019-11-21:
I had a look at the first option, see comments in #433, and there's no way: either we officially abandon libremap (removing the `ubus-lime-location` dependency from `lime-app`) or we'll always need LuCI.
The second option remains valid.
What's your opinion @spiccinini @nicopace @gmarcos87 @G10h4ck ?

spiccinini, 2019-11-21:
> I had a look at the first option, see comments in #433, and there's no way: either we officially abandon libremap (removing the `ubus-lime-location` dependency from `lime-app`) or we'll always need LuCI.
> The users will be able to include it selecting manually `ubus-lime-location` or deselecting it for having a firmware for low RAM/low flash memory devices.
> 
> The second option remains valid.
> What's your opinion @spiccinini @nicopace @gmarcos87 @G10h4ck ?

Yes I would go with the second one but modifying `uhttpd.main.index_page=lime_app_index.html,.......` (or using index_file it is not clear to me from the docs). This way we ship this other index with the limeapp and we don't have to change the paths. What do you think?

ilario, 2019-11-22:
> uhttpd.main.index_page=lime_app_index.html

Sounds like the best option so far!!

ilario, 2019-11-25:
> I would go with the second one but modifying uhttpd.main.index_page=lime_app_index.html

Tested, it works!

ilario, 2019-11-26:
Addressed in #651.
For users installing both full LuCI and lime-app interfaces (e.g. @digitigrafo) we **could** add a link somewhere for going from lime-app to the LuCI interface.

ilario, 2020-02-12:
Regression in #682 made this happen again.

ilario, 2020-02-12:
Fixed again by #690

ilario, 2020-08-28:
Seems that this is happening again, I don't know why and I didn't confirm personally: #752

spiccinini, 2020-08-28:
I can't reproduce it in 18.06.8

ilario, 2020-08-29:
@rallep71 can you do some more debugging?
I just tested an image compiled yesterday on 19.07-snapshot and the redirect works ok for me.
For example would be useful if from your computer you launch
`wget thisnode.info`
and paste here the output (instead of thisnode.info you can use the IP of your router if you prefer).

ilario, 2020-08-30:
The user reported that now the redirect works.

PR #651 (2019-11-27): lime-app installs the redirect to /app
Fixes #609 
The new index file used for redirection has been copied from LibreRouter's OpenWrt repository [here](https://github.com/LibreRouterOrg/openwrt/blob/d2282e29b0888dfa272149d845f48b202ec61b6f/files/www/index.html).
closed 2020-08-30 by ilario