libremesh/lime-packages #610: if the kernel needs to kill a process, trigger a reboot

suarezwilliamf, 2019-10-04:
it may happen that because the device is running out of memory the kernel, using omkiller, may want to kill some processes.
the libremesh experience depends on many processes, so we should monitor their uptime or reboot them directly.
(this is @nicopace by the way :) )

nicopace, 2019-10-08:
The reason why it is a reboot and not a service reboot, is that we don't know which service is the one that is eating the ram... and at that stage, we are late already for fixing it.
Also, we should think what happens if we reboot too many times... maybe the router could enter to a failsafe mode automatically... like at uboot... @spiccinini, what do you think ?

ilario, 2019-10-08:
We could implement something like: 
if (uptime > 24h and free RAM < 2 MB) then reboot

But if it is a process filling the RAM, and not the tmpfs (/tmp), the OOM killer should work, I saw it acting just once and it killed LuCI which was the correct process to kill...

ilario, 2023-03-02:
No solid solution and no interest in the last 4 years, closing

closed 2023-03-02 by ilario