libremesh/lime-packages #612: Homogenize adhoc_ssid between FirstBootWizard and lime-defaults-factory

ilario, 2019-10-04:
adhoc_ssid in FirstBootWizard code and in lime-defaults-factory should be identical and they are not.
See
https://github.com/libremesh/lime-packages/blob/e9efbea1003c8e333f5d34c220274d13d0ce5550/packages/first-boot-wizard/files/usr/lib/lua/firstbootwizard.lua#L283
and 
https://github.com/libremesh/lime-packages/blob/18c8aa59ec14bf8a2e82f72b0b761194d525213c/packages/lime-system/files/etc/config/lime-defaults-factory#L57

I can see no reason for having that line in FirstBootWizard code (and the next about ieee80211s_mesh_id), deleting it may fix.
Additionally, those two lines could break communities setting a custom adhoc_ssid and ieee80211s_mesh_id in their lime-defaults or network-profiles.

ilario, 2020-06-05:
The ssid has been homogenized sometimes in the past.
Anyway there are other settings that look different:

https://github.com/libremesh/lime-packages/blob/606ae4c6b02118571a3d312b6426d1c57e4e796c/packages/first-boot-wizard/files/usr/lib/lua/firstbootwizard.lua#L155

https://github.com/libremesh/lime-packages/blob/9d9837d2809c5fae1de3c3b804161a77d70fc785/packages/lime-system/files/etc/config/lime-defaults#L63-L66

https://github.com/libremesh/lime-packages/blob/606ae4c6b02118571a3d312b6426d1c57e4e796c/packages/first-boot-wizard/files/usr/lib/lua/firstbootwizard.lua#L149

https://github.com/libremesh/lime-packages/blob/9d9837d2809c5fae1de3c3b804161a77d70fc785/packages/lime-system/files/etc/config/lime-defaults#L54-L55

Can't FBW just take the values from lime-defaults config file instead of having hardcoded ones?

spiccinini, 2021-03-26:
Related to this issue https://github.com/libremesh/lime-packages/pull/859