libremesh/lime-packages #619: LiMe banner not always displayed

spiccinini, 2019-10-09:
Some installs show the libremesh banner and some don't.

ilario, 2019-10-09:
This is weeeeird!
That is implemented via [this file](https://github.com/libremesh/lime-packages/blob/00ae256396aa0ebdb0221d16eb4b8377062b69af/packages/lime-system/files/etc/uci-defaults/90_lime-banner) in lime-system.

ilario, 2019-12-04:
I saw this bug for the first time today, after using the `firstboot` command (which is *not* FirstBootWizard, it's an OpenWrt command for erasing the changes made since the firmware flashing).
When executing it, I see:
```
This will erase all settings and remove any installed packages. Are you sure? [N/y]
/dev/mtdblock4 is mounted as /overlay, only erasing files
```
and at the next reboot, the LibreMesh-specific part of the banner disappeared.
I suppose this is related to the fact that we are using a script in `/etc/uci-defaults`.
I checked whether other uci-defaults scripts were not effective after a `firstboot`:

`lime-system/files/etc/uci-defaults/89_lime-release` SEEMS OK
`lime-system/files/etc/uci-defaults/90_lime-banner` NOT GOOD
`lime-system/files/etc/uci-defaults/90_lime-customize-profile` OK
`lime-system/files/etc/uci-defaults/91_lime-config` NOT CHECKED
`lime-system/files/etc/uci-defaults/92_add-lime-repos` OBVIOUSLY FAILING as $LEDE_ARCH does not exist
`lime-app/files/95-lime-app-rpc-acl` OK
`lime-app/files/96-lime-app-index_page` OK
`shared-state-persist/files/etc/uci-defaults/50-set-automount` OK
`shared-state*/files/etc/uci-defaults/*` OK
`hotplug-initd-services/files/etc/uci-defaults/hotplug-initd-services` OK
`check-date-http/files/etc/uci-defaults/check-date-http-cron` OK
`ubus-lime-metrics/files/etc/uci-defaults/99-save-last-known-path-to-internet` OK

OK means that the result of these scripts was applied or persisted also after the `firstboot` command.

ilario, 2020-01-09:
> lime-system/files/etc/uci-defaults/90_lime-banner NOT GOOD

I suppose that also OpenWrt banner is created by a similar script and, for some weird reason, after the real first boot or after the firstboot command, the order of the two scripts is inverted?

>  lime-system/files/etc/uci-defaults/92_add-lime-repos FAILING when tested on builds based on OpenWrt 19.07-rc, see #275

Fixed by @spiccinini in #667

dangowrt, 2020-06-02:
`/etc/banner` is generated as build-time and part of base-files package in OpenWrt. I also observed that appending to a file doesn't work on recent version of overlayfs if it doesn't yet exists in /overlay/upper. this is very odd and probably a bug in overlayfs.
```
/rom/etc/uci-defaults/90_lime-banner 
/rom/etc/uci-defaults/90_lime-banner: line 3: can't create /etc/banner: Interrupted system call
```

PR #802 (2020-11-14): lime-banner use /etc/profile.d/ rather than /etc/banner
Fix #619
closed 2020-11-14 by spiccinini