libremesh/lime-packages #640: FBW overwrites my manually written configuration

ilario, 2019-11-13:
What happened:
I flashed a node with clean LibreMesh (using master branch code of a few weeks ago), set a root password, configured the node editing the `/etc/config/lime` file.
Then I opened `http://thisnode.info/app` (without specifying the /app suffix it fails due to #609) and the First Boot Wizard (FBW) pops up saying "Your router has not yet been configured".
Clicking Ok and going on with the configuration overwrites all the manually made configuration.

What I expected to happen:
If a root password is set, FBW should ask for login before proceeding.
If manual configuration is present, FBW should not open.

Risk:
Anyone (even random people on the internet, see #280 and [this issue reported 3 years ago](https://lists.libremesh.org/pipermail/lime-dev/2016-December/000821.html)) can break my router configuration, without even needing the root password that was set.

spiccinini, 2019-11-14:
Some part of this issue will be fixed with #635 as FBW will not overwrite any lime-node config but hostname. 

You are right about the risks

spiccinini, 2019-11-21:
I would say that this issue is fixed in part by #635 as now FBW won't overwrite your config in lime-node and also you can now (because of #639) use `option firstbootwizard_configured true` so that FBW should not run. @ilario  What do you think?

ilario, 2019-11-25:
> Some part of this issue will be fixed with #635 as FBW will not overwrite any lime-node config but hostname.
> now FBW won't overwrite your config in lime-node

I tested FBW with the presence of #635. I had some custom configuration in /etc/config/lime-node (configured just for avoiding using batman-adv and for having WPA2 on AP and APname), after running FBW wizard the custom configuration in /etc/config/lime-node disappeared!!

> and also you can now (because of #639) use option firstbootwizard_configured true so that FBW should not run.

This works :)
So, in order to avoid having FBW running, people need to have:

```
config lime 'system'
	option firstbootwizard_configured 'false'
```

in their `/etc/config/lime-node` AND run `lime-config` AND reboot.
No idea why but if I do not reboot the FBW ignores the new setting, but that's ok.

ilario, 2020-01-10:
Even if FBW erases all the configuration and it shouldn't, I think we can consider this closed.

We will have to stress on the fact that setting
```
config lime 'system'
	option firstbootwizard_configured 'false'
```
option is important for everyone changing configs by hand.

AndyMcSchopf, 2020-09-25:
> 
> 
> Even if FBW erases all the configuration and it shouldn't, I think we can consider this closed.
> 
> We will have to stress on the fact that setting
> 
> ```
> config lime 'system'
> 	option firstbootwizard_configured 'false'
> ```
> 
> option is important for everyone changing configs by hand.

by the way, shouldn't the option be "true" to be active?

ilario, 2020-09-30:
> > ```
> > config lime 'system'
> > 	option firstbootwizard_configured 'false'
> > ```
> by the way, shouldn't the option be "true" to be active?

I think that @spiccinini decided the phrasing.
For me it makes sense, more or less that option means: "has FBW already configured the system?" if you set to true is telling to FBW to not run as there is no need to configure anything anymore (either because it's already configured by FBW or by the user manually).

I'm going to re-open this issue as @germanferrero is addressing it, thanks!!! (see https://github.com/libremesh/lime-packages/issues/763#issuecomment-701005886)

ilario, 2020-10-31:
Should be fixed with https://github.com/libremesh/lime-app/pull/280

closed 2020-10-31 by ilario