libremesh/lime-packages #645: Update lime-proto-bgp BIRD dependencies

ilario, 2019-11-24:
With OpenWrt 19.07 the packages `bird4` and `bird6` does not exist anymore: they have been merged into a package named `bird1`.
A new version of that appeared with the name of `bird2`.

ilario, 2020-05-05:
@g10h4ck could you check this out please?

G10h4ck, 2020-05-06:
no time/means for that right now :-\

a-gave, 2020-06-29:
it seems that OpenWRT 19.07 keeps the previous packages bird4 and bird6
it has renamed them to bird1-ipv4 and bird1-ipv6

so it's temporary fixable if changing (at line 24)
./package/feeds/libremesh/lime-proto-bgp/Makefile 
with:
#  DEPENDS:=+bird4 +bird6 +lime-system +lua
  DEPENDS:=+bird1-ipv4 +bird1-ipv6 +lime-system +lua
 
NOT tested, but it should probably works, and this issue can be renamed to 
"Update lime-proto-bgp BIRD2 dependencies"

ilario, 2020-08-29:
> it seems that OpenWRT 19.07 keeps the previous packages bird4 and bird6
> it has renamed them to bird1-ipv4 and bird1-ipv6
> 
> so it's temporary fixable if changing (at line 24)
> https://github.com/libremesh/lime-packages/blob/master/packages/lime-proto-bgp/Makefile#L24
> with:
> `DEPENDS:=+bird1-ipv4 +bird1-ipv6 +lime-system +lua`

@a-gave your solution sounds like a great temporary fix. Can you prepare the pull request with this change? After this is updated the way you say, I would close this issue assuming that it's fixed.

rallep71, 2020-09-23:
Hello everybody,

I've changed that for me now, it works. 

please fix it in the Master

ilario, 2020-09-23:
@rallep71 do you mind doing a pull request with this change?

rallep71, 2020-09-23:
Hello,
i don't really understand this question, i think that the developers of libremesh are reading this. 

i have made this change in my site and it works, i think it should be in the code . Reason, other people who compile the code don't have this error message in make anymore.

and honestly i don't know much about the pull requests here at github

ilario, 2020-09-23:
> and honestly i don't know much about the pull requests here at github

If you want to try, here are some instructions (the easiest way I know), otherwise tell me and I'll do it, no problem at all :)

1) open the file you want to modify on Github: https://github.com/libremesh/lime-packages/blob/master/packages/lime-proto-bgp/Makefile#L24
2) click on the "Edit this file" button on the top right
3) do the edit
4) in the fields below, write a description for the commit (can be as easy as "lime-proto-bgp update dependencies")
5) click on "Commit changes"

rallep71, 2020-09-23:
hello ilario,
thank you for your answer.
i would like to ask you to do it for me right now, i have to read up on the committing first, thank you

ilario, 2020-09-23:
Done :D 
Check out #764, it will remove the warning for people compiling on 19.07 but it will add the warning for people still using 18.06.

PR #764 (2020-10-30): lime-proto-bgp depend on bird1-* packages from OpenWrt 19.07
Fix #645

This will remove the Warning (it is not an Error!!!) from the compilation with OpenWrt 19.07 but will add the warning to the compilation with OpenWrt 18.06. I have no idea how to write a dependency valid both for OpenWrt 18.06 and 19.07 without using conditional dependencies which are evil (see #331 #673 ).

Here you can find the [Bird Makefile in 18.06](https://github.com/openwrt-routing/packages/blob/openwrt-18.06/bird/Makefile) and the [Bird1 Makefile in 19.07](https://github.com/openwrt-routing/packages/blob/openwrt-19. [... cut]
closed 2020-10-30 by G10h4ck