libremesh/lime-packages #647: HT40 not accepted as channel width

ilario, 2019-11-24:
Checking `logread` I noticed an error related to the 5 GHz channel width used: HT40 is not an accepted value (already reported in #630).

```
daemon.notice netifd: radio1 (2614): Usage:	iw [options] dev <devname> set channel <channel> [NOHT|HT20|HT40+|HT40-|5MHz|10MHz|80MHz]
daemon.notice netifd: radio1 (2614): Options:
daemon.notice netifd: radio1 (2614): 	--debug		enable netlink debugging
user.notice mac80211: Failed command: iw dev wlan1-mesh set channel 48 HT40
daemon.err hostapd: Configuration file: /var/run/hostapd-phy1.conf
```

This is in contrast with the [documentation on OpenWrt website here](https://openwrt.org/docs/guide-user/network/wifi/basic#common_options).

I thought that replacing `HT40` with `HT40-` was a good idea (see [the attempt here](https://github.com/libremesh/lime-packages/compare/330ae4ca17d5f0b490f12dd280990d94a3ecca2c...ilario:ht40)) but actually this would work for channel 48 but not for channel 157.
Check out the allowed HT40+ or HT40- [on this website](https://www.digi.com/resources/documentation/digidocs/90001394/references/r_5ghz_h20_and_ht40_chanl.htm).

One possible solution is to copy the system employed [in lime-smart-wifi here](https://github.com/libremesh/lime-packages/blob/ede4b03a4df3af5c9b3be1545e98099c728e6ba9/packages/lime-smart-wifi/src/lime-smart-wifi.lua#L61-L85) and use it in lime-system's wireless.lua?
Otherwise we could accept a list of HT* related to the list of 5 GHz channels???

I tried replacing the default iw package with the iw-full package but HT40 (without +/-) is still not accepted.

dangowrt, 2019-11-24:
I'm trying to understand how this call to `iw` is happening. In OpenWrt, we are automatically replacing `HT40` with `HT40-` or `HT40+` depending on band and channel, see
https://github.com/openwrt/openwrt/blob/master/package/kernel/mac80211/files/lib/netifd/wireless/mac80211.sh#L726
and
https://github.com/openwrt/openwrt/blob/master/package/kernel/mac80211/files/lib/netifd/wireless/mac80211.sh#L631

However, this is apparently not the case here:
https://github.com/openwrt/openwrt/blob/master/package/kernel/mac80211/files/lib/netifd/wireless/mac80211.sh#L554

Given that the scheme is identical for mesh and ad-hoc mode, probably the best would be to have a shell-function for that (currently redundant) functionality and use it in L554 as well.

Do you need help preparing a patch for that?

spiccinini, 2019-11-25:
@dangowrt Something like this looks ok to you https://github.com/spiccinini/openwrt/commit/b4dca77e634796ac273c759688c81b938e4d2856 ?

ilario, 2019-11-29:
Just for making it clear: @spiccinini filed a pull request to OpenWrt here: https://github.com/openwrt/openwrt/pull/2573
@dangowrt do you think it's likely that we'll have this into OpenWrt 19.07?

dangowrt, 2019-11-29:
I've picked that into master and as it is a fix, I'll cherry-pick it
to 19.07 in a couple of days once more people have tested it.

ilario, 2019-12-03:
I heard that the next release will be based on OpenWrt 18.06, so the fix included in 19.07 will not be there :/

ilario, 2020-01-10:
Now that [LibreRouterOS has been released](https://lists.libremesh.org/pipermail/lime-users/2020-January/001558.html) and is based on OpenWrt 18.06, I see no reason for not having the next LibreMesh release based on OpenWrt 19.07. In the latter, this issue is already solved. Closing.

spiccinini, 2020-03-23:
@ilario I don't understand why you reopened? From LibreMesh we can't fix it. Maybe we can have a known issues and add it there?

ilario, 2020-03-23:
> I don't understand why you reopened? 

Just for remembering that it's a known bug.¡, in case we make a release based on OpenWrt18.06

> From LibreMesh we can't fix it.

I suppose we could but it's not worth to do it. I was not proposing to fix this on our side.

> Maybe we can have a known issues and add it there?

Yes please! This would go in the direction of making a release (based on OpenWrt18.06). Suggestions on where to start working on this? For example an Etherpad connected to the Riot/Matrix channel (#libremesh-dev:matrix.guifi.net) or a git repo here?

ilario, 2021-11-28:
I just tried using HT40 with channel 151 (good for short distance links, I had to set up a link between different floors of the same condo in open air) on LibreMesh master compiled with OpenWrt 19.07.8 and I still got the message:

```
user.notice mac80211: Failed command: iw dev wlan1-mesh set channel 151 HT40+
```

I had to change the channel to 153 and then it worked.
So maybe @spiccinini commit was not included in 19.07.8?

ilario, 2021-11-29:
>  So maybe @spiccinini commit was not included in 19.07.8?

Just checked: that commit is included in 19.07.8, so maybe it didn't fix?
https://github.com/openwrt/openwrt/commits/v19.07.8/package/kernel/mac80211/files/lib/netifd/wireless/mac80211.sh

ilario, 2023-04-07:
Just confirmed compiling on OpenWrt 22.03.3 using a YouHua WR1200JS:

using channel 48 worked:

```
# iw dev wlan1-ap info
Interface wlan1-ap
	ifindex 35
	wdev 0x100000005
	addr d6:5f:25:eb:7f:66
	ssid LibreMesh.org
	type AP
	wiphy 1
	channel 48 (5240 MHz), width: 40 MHz, center1: 5230 MHz
	txpower 20.00 dBm
	multicast TXQ:
		qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcol	tx-bytes	tx-packets
		0	0	0	0	0	0	0	0		0
```

using channel 157 did not work:

```
# iw dev wlan1-ap info
Interface wlan1-ap
	ifindex 45
	wdev 0x100000008
	addr d6:5f:25:eb:7f:66
	type AP
	wiphy 1
	txpower 20.00 dBm
	multicast TXQ:
		qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcol	tx-bytes	tx-packets
		0	0	0	0	0	0	0	0		0
```

It is not a problem of regulatory domain:

```
# iw reg get
global
country 00: DFS-UNSET
	(755 - 928 @ 2), (N/A, 20), (N/A), PASSIVE-SCAN
	(2402 - 2472 @ 40), (N/A, 20), (N/A)
	(2457 - 2482 @ 20), (N/A, 20), (N/A), AUTO-BW, PASSIVE-SCAN
	(2474 - 2494 @ 20), (N/A, 20), (N/A), NO-OFDM, PASSIVE-SCAN
	(5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW
	(5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW, PASSIVE-SCAN
	(5490 - 5730 @ 160), (N/A, 20), (0 ms), DFS, PASSIVE-SCAN
	(5735 - 5835 @ 80), (N/A, 20), (N/A), PASSIVE-SCAN
	(57240 - 63720 @ 2160), (N/A, 0), (N/A)
```

ilario, 2023-04-08:
We just tried HT80 on a Plasma Cloud PA1200 and failed using channel 36, 40, 42 and 44.

ilario, 2023-04-11:
@dangowrt @spiccinini can you help on this?
It would be great to use HT80 for the network inside Calafou, as they have a gigabit internet connection and no wifi noise from the surroundings.
Thanksssss!

dangowrt, 2023-04-11:
Only very few channels will work for HT80, as regulation doesn't allow overlap. Ie. you have to specify the lower 20 MHz channel (and *not* the center channel!), and only 36, 52, 100, 116, 132 and 149 will work inside EU regulatory domain for HT80. For indoor AP application use DFS channel (52, 100, 116, 132) and avoid SDR band which has low permitted TX power.
Are you using patched regulatory domain? If so, there might be a problem with that...

ilario, 2023-04-12:
Sorry for the fuss, my fault. I did not read the documentation, where it clearly say that I should use `VHT80` rather than `HT80`.
Now it works!!!!! Thanks!

ilario, 2024-11-10:
From the comments above, this seems fixed.

closed 2024-11-10 by ilario