libremesh/lime-packages #649: Batman-adv configuration needs to be adapted to upstream changes

ilario, 2019-11-25:
When compiling LibreMesh on top of the OpenWrt's openwrt-19.07 branch, I had several errors related to Batman-adv interfaces:

```
daemon.notice netifd: Interface 'lm_net_batadv_dummy_if' is setting up now
daemon.notice netifd: Interface 'lm_net_eth0_2_batadv_if' is now down
daemon.notice netifd: Interface 'lm_net_eth0_2_batadv_if' is setting up now
daemon.notice netifd: Interface 'lm_net_wlan0_mesh_batadv_if' is now down
daemon.notice netifd: Interface 'lm_net_wlan0_mesh_batadv_if' is setting up now
daemon.notice netifd: lm_net_wlan1_mesh_batadv_if (22621): Command failed: Invalid argument
daemon.notice netifd: Interface 'lm_net_wlan1_mesh_batadv_if' is now down
daemon.notice netifd: Interface 'lm_net_wlan1_mesh_batadv_if' is setting up now
daemon.notice netifd: lm_net_batadv_dummy_if (22662): Command failed: Invalid argument
daemon.notice netifd: lm_net_eth0_2_batadv_if (22671): Command failed: Invalid argument
```

Asking for help on Batman-adv IRC seems that the configuration has to be updated due to this commit as documented in the same commit:

https://github.com/openwrt-routing/packages/commit/54af5a209e0a0a75b5eb712c0ca8056e66de02c0

Checking on `ubus monitor` the ubus commands causing the error seems to be:

```
<- 6c63e636 #a20df0f7         invoke: {"objid":-1576144649,"method":"notify_proto","data":{"action":0,"ifname":"lm_net_wlan1_mesh_batadv_if","link-up":true,"keep":false,"interface":"lm_net_wlan1_mesh_batadv_if"}}
-> b84bd45b #6c63e636         invoke: {"objid":-1576144649,"method":"notify_proto","data":{"action":0,"ifname":"lm_net_wlan1_mesh_batadv_if","link-up":true,"keep":false,"interface":"lm_net_wlan1_mesh_batadv_if"},"user":"root","group":"root"}
<- b84bd45b #6c63e636         status: {"status":2,"objid":-1576144649}
-> 6c63e636 #a20df0f7         status: {"status":2,"objid":-1576144649}
```

ilario, 2019-11-25:
On Batman-adv IRC I received this comments:

```
<ecsv> two odd things here - why is the status here 2? didn't you say " Invalid argument" (aka 22) before?
<ecsv> and then it it is odd that ifname is lm_net_wlan0_mesh_batadv_if. that is much too long for ifname (which is limited to 16 characters in the kernel). i was expecting something more like wlan0 or mesh0 or something like this
<ecsv> and the wireless scripts of netifd/mac80211 should actually take care of filling this value in
<ecsv> so if you have an ifname in /etc/config/wireless for an interface which belongs to an wifi-iface from /etc/config/wireless than you are most likely doing something wrong
<ecsv> or did you call the interface with proto batadv in /etc/config/network lm_net_wlan0_mesh_batadv_if? if this is the case then it is also wrong.
<ecsv> the section name for the interface with the proto 'batadv' will be the name of the batadv interface (normally called bat0)
<ilario_> there also plenty of other lines with status 0, but these ones where the ones mentioning the same interfaces as in the errors from logread...
<ilario_> the names are defined in /etc/config/network, yes, I uploaded the network file here (it is large, due to LibreMesh configuring plenty of interfaces for doing various stuff): https://uz.sns.it/~ilario/batadv-20191125/
<ecsv> 0 means no error
<ecsv> the batman-adv file shouldn't be there
<ecsv> it is now all part of the batadv interface
<ecsv> lm_net_batadv_dummy_if should not have the proto batadv but batadv_hardif
<ecsv> same for lm_net_wlan0_mesh_batadv_if
<ecsv> same for lm_net_wlan1_mesh_batadv_if
<ecsv> same for lm_net_eth0_2_batadv_if
```

aparcar, 2020-05-05:
@dangowrt I never used batman adv but maybe you can help out here to get things working on 19.07

dangowrt, 2020-05-29:
Deployed the version now in master (ie. including the compatibility code added) on 19.07.3 now and works very nice.

closed 2020-05-29 by dangowrt